kasbert/OS-X-SAT-SMART-Driver #36: OUSBFamily was not able to enumerate the device

paullinford, 2015-03-29:
after installing driver, drive is ultimately ejected

3/29/15 7:28:06.000 AM kernel[0]: USBF:	283.561	The IOUSBFamily is having trouble enumerating a USB device that has been plugged in.  It will keep retrying.  (Port 5 of Hub at 0xfd000000)
3/29/15 7:28:12.000 AM kernel[0]: USBF:	289.532	AppleUSBEHCI::Found a transaction which hasn't moved in 5000 milliseconds on bus 0xfd, timing out! (Addr: 0, EP: 0)
3/29/15 7:28:18.000 AM kernel[0]: USBF:	295.535	AppleUSBEHCI::Found a transaction which hasn't moved in 5000 milliseconds on bus 0xfd, timing out! (Addr: 0, EP: 0)
3/29/15 7:28:24.000 AM kernel[0]: USBF:	301.538	AppleUSBEHCI::Found a transaction which hasn't moved in 5000 milliseconds on bus 0xfd, timing out! (Addr: 0, EP: 0)
3/29/15 7:28:30.000 AM kernel[0]: USBF:	307.539	AppleUSBEHCI::Found a transaction which hasn't moved in 5000 milliseconds on bus 0xfd, timing out! (Addr: 0, EP: 0)
3/29/15 7:28:36.000 AM kernel[0]: USBF:	313.541	AppleUSBEHCI::Found a transaction which hasn't moved in 5000 milliseconds on bus 0xfd, timing out! (Addr: 0, EP: 0)
3/29/15 7:28:42.000 AM kernel[0]: USBF:	319.544	AppleUSBEHCI::Found a transaction which hasn't moved in 5000 milliseconds on bus 0xfd, timing out! (Addr: 0, EP: 0)
3/29/15 7:28:42.000 AM kernel[0]: USBF:	320. 78	The IOUSBFamily was not able to enumerate the device (Port 5 of Hub at 0xfd000000).

### SYSTEM INFORMATION ###
Report Timestamp                     : March 29, 2015 7:29:00 AM PDT
Report Timestamp (ISO 8601 format)   : 2015-03-29T07:29:00

Application Name                     : DriveDx
Application Version                  : 1.3.0.444
App SubBuild                         : 0
Application Edition                  : 1
DriveDx Knowledge Base Revision      : 1/1
Computer Model                       : MacPro5,1
OS Name                              : Mac OS X
OS Version                           : 10.8.5
OS Build                             : 12F2501
OS Kernel Version                    : Darwin 12.6.0

Same device, same issue on MacPro4,1

alvarnell, 2015-03-30:
You certainly haven�t given kasbert much to go on here.  Not surprising that it isn�t platform related, but I suspect he�s going to need to know:

- What version of the driver you installed.

- What is the make and model of the device.

- Where you have it plugged in, i.e. the computer, the keyboard, a usb hub?  If not an Apple keyboard or a third party hub, what make and model.

Also did you check <http://sourceforge.net/apps/trac/smartmontools/wiki/Supported_USB-Devices>?  If the options column has anyting, but "-d sat", the enclosure does not support SAT and therefore does not work with this driver.

-Al-

kasbert, 2015-04-12:
Would you try version 0.11-Debug, please ?
If needed, allow unsigned kernel extensions with command:
  sudo nvram boot-args="kext-dev-mode=1

I have occasionally seen the enumeration failing with my LaCie enclosure, but haven't been able to reproduce the error. I suspect that the enclosure is not ready for identify command during the driver initialization, so I delayed the identify a bit.

paullinford, 2015-04-12:
Hi Jarkko
I installed SATSMARTDriver-0.11-Debug.dmg
Console now shows when it is attached
4/12/15 10:33:09.000 AM kernel[0]: USBMSC Identifier (non-unique):
00027a2400004798 0x3873 0x924 0x0, 2

The bridge device doesn't seem to be SAT SMART anyway. No info via USB or
FW800 using DriveDX.

Any info I can give you or do I consider the bridge device too old?

Paul

kasbert, 2015-04-13:
So now the drive works (does not fail to enumerate), but the SMART info is not available and it is causing problems in starting ?

Resetting pram might help, since the kernel should no longer allow the driver to load.

It's possible that the bridge does not support SAT commands, because it is a raid one. Does the driver show drive model correctly ?

paullinford, 2015-04-13:
I've removed it. Guess I can install and look, then remove before starting.

> So now the drive works (does not fail to enumerate), but the SMART info is
> not available and it is causing problems in starting ?
Yes, SMART status not supported, and would not complete startup.

> Resetting pram might help, since the kernel should no longer allow the
> driver to load.
I booted to recovery and ran modified terminal commands to remove the
driver and plugin.

> It's possible that the bridge does not support SAT commands, because it is
> a raid one. Does the driver show drive model correctly ?
Yes it is capable of raid via a jumper I think. It is currently JBOD, and
both channels have hot swap bays, but at the moment only one drive is
inserted.