kasbert/OS-X-SAT-SMART-Driver #37: driver causes apps that check S.M.A.R.T. status to crash

core-code, 2015-04-20:
we received several crash reports via iTunesConnect about our app crashing when checking the SMART status. the crash happens right in your driver code in response to our app calling IODestroyPlugInInterface()

the crash seems to be present in versions 0.5 through 0.10

crash logs:

http://dl.dropboxusercontent.com/u/7221986/_crash_kasbert.zip

kasbert, 2015-04-22:
I have no idea. My only guess is that opening the client fails, because there can be only one client at the same time. I guess you haven't been able to reproduce the problem either. I suggest adding some logging to the code containing IODestroyPlugInInterface. Then use other tools to reserve the driver.

core-code, 2015-04-22:
hello kasbert

> My only guess is that opening the client fails, because there can be only one client at the same time

no the crash has to have another cause. our code is based directly on Apple Sample Code for S.M.A.R.T. status checking and exits when IOCreatePlugInInterfaceForService() fails. i just tested this by starting two instances of our app in the debugger.
here is the code in detail:
https://gist.github.com/core-code/79505e6782f0e3fda2ac

to get to the IODestroyPlugInInterface(), IOCreatePlugInInterfaceForService() must have returned kIOReturnSuccess

no, i've never been able to reproduce it, but as i said its happening on customer macs

emory, 2015-06-08:
FWIW I've seen this behavior and similar interaction, but I think I was leaning towards GenericXHCI as the more likely source in my case — any chance your customers are using that extension, @core-code

core-code, 2015-06-08:
no, as you can see in the crash logs, only 'OS-X-SAT-SMART-Driver' is present and the crash is happening right in its code

core-code, 2015-08-03:
i had to change the location of the crash logs

they are here now:

http://pastebin.com/RsUR6NXU

p6v9um5wjm6jkmqw, 2015-08-09:
Guys, please forgive me for posting this here but I don't know where else to get in touch with the developers for this driver. Today I installed the SAT SMART driver (or rather, DriveDX installed it for me) and caused a bit of a problem. DriveDX worked fine, and your driver allowed me to check my external drives, but I also use Chameleon SSD Optimizer to enable Trim on my aftermarket SSD, and launching it after using DriveDX caused my computer to hard restart. When it came back up one of my external drives was not mounting. Disk Utility would not fix it, so I had to purchase Disk Warrior, which fixed it instantly. First thing I did was remove the SAT SMART driver, and launched Chameleon. It came up without a hitch this time. Just thought this would be good information to pass along. Sorry if this isn't the place.

core-code, 2016-10-17:
its quite unnerving that people think that our SMARTReporter is crashing while its really the "OS-X-SAT-SMART-Driver" thats crashing.

i've thought about refusing to launch when the driver is installed, since these crash reports seem to be quite frequent.

could you at least turn on debug symbols in your official binaries so that the generated debug logs could point to the source of the problem?

kasbert, 2016-10-18:
I'll try to find time to look at the issue.

JanX2, 2016-10-18:
What about a donation @core-code? :)

core-code, 2016-10-18:
is there a way to donate to this project or this bug report? bountysource?