kasbert/OS-X-SAT-SMART-Driver #38: code signing issues and external devices not loading

paulbeard, 2015-05-10:
I can install from the packaged installers but the external devices don't show up. 

Or I can build from source (which has its own problems). 
`echo VERSION 0.11`
`VERSION 0.11`
`echo PKG SATSMARTDriver-0.11-Debug.pkg`
`PKG SATSMARTDriver-0.11-Debug.pkg`
`rm -rf Root SATSMARTDriver-0.11-Debug.pkg`
`mkdir -p Root/System/Library/Extensions/`
`ditto --rsrc SATSMARTDriver/build/Debug/SATSMARTDriver.kext Root/System/Library/Extensions/SATSMARTDriver.kext`
`ditto --rsrc SATSMARTDriver/build/Debug/SATSMARTLib.plugin Root/System/Library/Extensions/SATSMARTLib.plugin`
`/Developer/Applications/Utilities/PackageMaker.app/Contents/MacOS/PackageMaker --doc SATSMARTDriver.pmdoc --out SATSMARTDriver-0.11-Debug.pkg`
`make: /Developer/Applications/Utilities/PackageMaker.app/Contents/MacOS/PackageMaker: No such file or directory`
`make: *** [pkg] Error 1`

Installing it leaves me with this: 
`$ sudo make install `
`ioreg -r -c IOSCSIPeripheralDeviceType00 -l | grep "BSD Name" | cut -d'"' -f4 | while read a; do diskutil unmountDisk "$a" || exit 1; done`
`Unmount of all volumes on disk1 was successful`
`Unmount of all volumes on disk1 was successful`
`Unmount of all volumes on disk1 was successful`
`sudo kextunload -v -b fi.dungeon.driver.SATSMARTDriver`
`(kernel) Kext fi.dungeon.driver.SATSMARTDriver not found for unload request.`
`Failed to unload fi.dungeon.driver.SATSMARTDriver - (libkern/kext) not found.`
`make: [unload] Error 3 (ignored)`
`sudo kextunload -v -b org.dungeon.driver.SATSMARTDriver`
`(kernel) Kext org.dungeon.driver.SATSMARTDriver not found for unload request.`
`Failed to unload org.dungeon.driver.SATSMARTDriver - (libkern/kext) not found.`
`make: [unload] Error 3 (ignored)`
`sudo cp -R SATSMARTDriver/build/Debug/SATSMARTLib.plugin /System/Library/Extensions`
`sudo rm -rf /tmp/SATSMARTDriver.kext`
`sudo cp -R SATSMARTDriver/build/Debug/SATSMARTDriver.kext /tmp`
`sync`
`sudo kextutil -t /tmp/SATSMARTDriver.kext`
`Cache file /System/Library/Caches/com.apple.kext.caches/Directories/System/Library/Extensions/KextIdentifiers.plist.gz is out of date; not using.`
`Notice: /tmp/SATSMARTDriver.kext has debug properties set.`
`Diagnostics for /tmp/SATSMARTDriver.kext:`
`Code Signing Failure: not code signed`
`kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/tmp/SATSMARTDriver.kext"`

alvarnell, 2015-05-10:
Only versions 0.8 and 0.9 have been whitelisted by Apple to work with Yosemite (OS X 10.10.x).

From the Read Me:

> Note for Yosemite (Mac OS X 10.10) users:
> Try first version 0.8. If it works for you, then just use it.
> If not or you want to try newer version or participate on development, 
> then you must allow unsigned kernel extensions with command:
>   sudo nvram boot-args="kext-dev-mode=1"

Doing so will allow all kext files, so be careful about any others you may install. Re-installing the OS will likely reset this, so you will need to remember to do it each time.

-Al-

-Al-
-- 
Al Varnell
Mountain View, CA

paulbeard, 2015-05-10:
> On May 9, 2015, at 6:18 PM, Al Varnell <notifications@github.com> wrote:
> 
> Only versions 0.8 and 0.9 have been whitelisted by Apple to work with Yosemite (OS X 10.10.x).

Well, none of it seems to work for me. Not sure what I have done wrong. The installers don’t seem to work (kextstat doesn’t show the kernel extensions as loaded). I think my problem is too many parts/choices. 

Trying 0.8 again now. 

It installed, per the installer, but wasn’t there in kextstat. I used kextload on it, confirmed it was there, but smartctl doesn’t see it. 

Oh, well. Thanks, Apple…

kasbert, 2015-05-11:
Was there anything in the kernel log ? 
What kind of enclosure you have ?

paulbeard, 2015-05-11:
> On May 10, 2015, at 9:44 PM, Jarkko Sonninen <notifications@github.com> wrote:
> 
> Was there anything in the kernel log ? 

Nothing of note. 

> What kind of enclosure you have ?

A jmicron that used to work with older versions. 

My internal drive (recently replaced) won't allow SMART to be set permanently which is likely unrelated but irritating.

kasbert, 2015-05-11:
You mean older than 0.8 ?
What are the USB vendor and product ids ?

paulbeard, 2015-05-11:
> On May 10, 2015, at 11:00 PM, Jarkko Sonninen <notifications@github.com> wrote:
> 
> You mean older than 0.8 ?
> 
I can’t recall. It was a while ago. 
> What are the USB vendor and product ids ?
> 

DMI USB2.0 Storage:

  Product ID:	0xb157
  Vendor ID:	0x0c0b  (Dura Micro, Inc.)
  Version:	0.00
  Serial Number:	201003020572
  Speed:	Up to 480 Mb/sec
  Manufacturer:	DMI
  Location ID:	0xfa130000 / 7
  Current Available (mA):	500
  Current Required (mA):	2
  Capacity:	1 TB (1,000,204,886,016 bytes)
  Removable Media:	Yes
  Detachable Drive:	Yes
  BSD Name:	disk1
  Partition Map Type:	GPT (GUID Partition Table)
  S.M.A.R.T. status:	Not Supported
  Volumes:
EFI:
  Capacity:	209.7 MB (209,715,200 bytes)
  BSD Name:	disk1s1
  Content:	EFI
  Volume UUID:	0E239BC6-F960-3107-89CF-1C97F78BB46B
Untitled:
  Capacity:	999.86 GB (999,860,912,128 bytes)
  Available:	999.2 GB (999,197,302,784 bytes)
  Writable:	Yes
  File System:	Journaled HFS+
  BSD Name:	disk1s2
  Mount Point:	/Volumes/Untitled
  Content:	Apple_HFS
  Volume UUID:	6D9C50C5-FF5E-3329-A6BB-7737B83E0E74

kasbert, 2015-05-11:
Smartmontools lists another device with nearly same ids (0c0b:b159) as a "sunplus" type. Sunplus ata-passthrough command is different compared to sat or jmicron. If yours is similar, smart reporting shouldn't have worked ever.

kasbert, 2015-05-31:
Would you please try the version "0.11-Debug" ?

1. Download and run "smart_status" command before installation. It should show something like 
```
ID              : 0x100000228
Class           : com_LaCie_ScsiType00
Device File     : disk1
SMART Query     : FAILED, disk1: kIOReturnUnsupported (e00002c7)

ID              : 0x100000219
Class           : IOSCSIPeripheralDeviceType05
SMART Query     : FAILED, kIOReturnUnsupported (e00002c7)

ID              : 0x1000001d9
Class           : AppleAHCIDiskDriver
Model           : INTEL SSDSA2BW120G3                     
Revision        : 4PC10362
Device File     : disk0
SMART Query     : ok
SMART Read      : ok
SMART Status    : OK
```

The Class tells, which driver handles the device. IOSCSIPeripheralDeviceType05 is a DVD drive.

2. Install the driver. Now the driver won't try to autodetect the enclosure, if it is not known. 

3. Run "smart_status" again. PassThroughMode should be none.
```
ID              : 0x100000228
Class           : fi_dungeon_driver_IOSATDriver
Enclosure Name  : CROSU2 SATA Docking Station
SMART Capable   : true
PassThroughMode : none
Device File     : disk1
SMART Query     : ok
disk1: kIOReturnInvalid (e0000001)

ID              : 0x10000021a
Class           : IOSCSIPeripheralDeviceType05
SMART Query     : FAILED, kIOReturnUnsupported (e00002c7)

ID              : 0x1000001c7
Class           : AppleAHCIDiskDriver
Model           : INTEL SSDSA2BW120G3                     
Revision        : 4PC10362
Device File     : disk0
SMART Query     : ok
SMART Read      : ok
SMART Status    : OK
```

4. Set passthrough mode with "smart_status ID sunplus". Pick the ID from previous command output. Try also "smart_status ID jmicron"
```
$ smart_status 0x100000228 sat12
ID              : 0x100000228
Class           : fi_dungeon_driver_IOSATDriver
Set PassThroughMode = sat12
Enclosure Name  : CROSU2 SATA Docking Station
SMART Capable   : true
PassThroughMode : sat12
Product Name    : WDC WD15EARS-00MVWB0
Model           : WDC WD15EARS-00MVWB0
Device File     : disk1
SMART Query     : ok
SMART Read      : ok
SMART Status    : OK
```