kasbert/OS-X-SAT-SMART-Driver #39: Which version works under OS X El Capitan 10.11.x?

macuserguru, 2015-08-09:
Have anyone tested OS X SAT SMART driver under OS X El Capitan 10.11.x?
Wich version works under OS X El Capitan 10.11.x without disable the new OS X El Capitan security features?
Or has Apple included the SMART driver for external hard disks self now in El Capitan?

Under Yosemite 10.10.4 I use 0.8 and it works well.

alvarnell, 2015-08-09:
Everybody using OS X 10.11 is under NDA and would not be able to discuss it here. If you really want to know, sign up for the Public Beta. But of course a lot could change between now and Fall release, so even if a particular version worked today it might not tomorrow.

-Al-

On Sun, Aug 09, 2015 at 01:24 AM, macuserguru wrote:
> Have anyone tested OS X SAT SMART driver under OS X El Capitan 10.11.x?
> Wich version works under OS X El Capitan 10.11.x without disable the new OS X El Capitan security features?
> Or has Apple [...]

casey1234, 2015-09-14:
Hi! 

I am using OSX El Capitan GM Candidate (10.11.0) and unfortunately, the SMART Driver for the app "DriveDx" does not work anymore. 
Will there be a near future update which will work fine under El Capitan? Would be awesome!!

Greetings from Germany!

macuserguru, 2015-09-14:
Version 0.8 don't work under El Capitan?

casey1234, 2015-09-14:
Unfortunately not! We really need an update so that we can still receive the data of our USB devices. I really hope there will be an update soon.

macuserguru, 2015-09-14:
If the developer needs an Apple signature and must be an registered Mac developer - I suggest to make this driver as shareware. I had no problem to buy it.

casey1234, 2015-09-14:
Same here, it's just too good to be left without an update!

alvarnell, 2015-09-14:
Since DriveDX and SMART Utility are paid apps, I would think it would be in there best interest to keep it working by covering any expenses involved.

Sent from Janet's iPad

-Al-

On Sep 14, 2015, at 5:56 AM, casey1234 <notifications@github.com> wrote:
> Hi!
> 
> I am using OSX El Capitan GM Candidate (10.11.0) and unfortunately, the SMART Driver for the app "DriveDx" does not work anymore. 
> Will there be a near future update which will work fine under El Capitan? Would be awesome!!
> 
> Greetings from Germany!
>

macuserguru, 2015-09-15:
I use DriveDX, but DriveDX needs this SMART driver too. Drive-DX offers a SMART driver download.
A co-working would be the best way.

alvarnell, 2015-09-15:
Technically, the SATSMARTDriver-0.8.pkg installer is built into DriveDX so it doesn’t need to be downloaded, but the end result is the same.  SMART Utility works the same way.  I assume they must have permission to do so and would hope that kasbert has been adequately compensated for it.

-Al-

-Al-
-- 
Al Varnell
Mountain View, CA

kasbert, 2015-09-15:
No, they haven't asked and they have a right to just grab the driver. And I don't have any problem with that. On the other hand I am not obliged to help them or their customers. There are all kinds of hardware out there and solving problems is a tedious job.

casey1234, 2015-09-15:
So you will not update the SMART Driver for El Capitan 10.11? :-(

kasbert, 2015-09-15:
I don't have El Capitan. I'll try updating the driver, when (if) I upgrade to El Capitan. Currenly my Mac is on production use (living room) and I don't want to mess it with beta versions. Users (kids) would complain about downtimes.

casey1234, 2015-09-15:
The current status is GoldMaster, so it is very close to the final version, but I respect your choice. The final will officially be released on September 30th, I hope you are going to update your OS soon after, so that we all can benefit from an update in the near future. If necessary make it shareway, I would not have any problem with paying you for the great work.

Greetings from Germany

RoasAnne, 2015-09-15:
Stop emailing me please

kasbert, 2015-09-16:
RoasAnne: browse to https://github.com/kasbert/OS-X-SAT-SMART-Driver  and click "Unwatch" and "Ignoring"

jaffacakerhubarb, 2015-09-29:
Just for the record, the GM of El Capitan works perfectly with the latest version — provided you have System Integrity Protection (SIP) disabled. I've been using it since its public beta release and have had no issues. Some background from the just-release Ars Technica review:

> Going hand-in-hand with the runtime protection is mandatory signing for kernel extensions. As of El Cap, the methods for working around KEXT signing (like using sudo nvram boot-args='kext-dev-mode=1') have been removed, and KEXT signing enforcement is now under SIP control. El Cap will refuse to load any kernel extensions that aren’t signed by a valid Apple developer ID.

As far as I know, that just requires a paid developer account.

macuserguru, 2015-09-30:
Off topic.
Whats the cost for a developer account?

casey1234, 2015-09-30:
@jaffacakerhubarb:

That means, it just works with a developer account?

jaffacakerhubarb, 2015-09-30:
To clarify: the developer would need to be part of the Mac/iOS developer program, then they can sign their apps/kexts. I believe this is $99 USD per year. This gives apple the ability to revoke developer IDs if they turn out to be somehow malicious or in violation of guidelines.

macuserguru, 2015-09-30:
I don't want to disable the SIP (rootless) in El Capitan - but I want to 'listen' for SMART messages from external hard disks. Which way gives for?

jaffacakerhubarb, 2015-09-30:
So kasbert would need to sign up to the Mac developer program (assuming he isn't already) and sign the SATsmartdriver. Then it would happily run under El Capitan, as far as I understand it.

This is based on my understanding of why it works with SIP disabled. It's possible, however, that it attaches itself to a running process or violates some other new precept, in which case it won't work even if signed. The Ars Technica review explains in more detail, (although the appropriate WWDC session video is probably the authority):

http://arstechnica.com/apple/2015/09/os-x-10-11-el-capitan-the-ars-technica-review/8/#h1

JanX2, 2015-09-30:
I have a dev account and would consider signing the kext.

alvarnell, 2015-09-30:
Now that El Camino has been released I can tell you that using OS X 10.11 I have been able to read the S.M.A.R.T. data from all the external drives I've tried without having the OS-X-SMART-SAT-driver extension installed, so there may be only limited need for it in the future.

macuserguru, 2015-10-01:
Have El Capitano the SMART-SAT driver for external USB, Firewire, TB, etc. disks now included self?

If yes - a very good message.

off topic -  I hope the tool HDAPM works with El Capitano, I need them for my WD blue 1 TB HD too (extend his lifetime).

http://mckinlay.net.nz/hdapm/
https://discussions.apple.com/message/6986177
http://forums.appleinsider.com/t/155256/hdapm
http://forums.macrumors.com/threads/hdapm-question.601219/

casey1234, 2015-10-01:
@alvarnell:
Does it also work with external USB drives? 
What App do you use for reading the SMART data?

alvarnell, 2015-10-02:
It should, but I don’t recall that I tried. I use Thunderbolt 1 and FireWire 800 almost all the time.

I use SMART Utility, DriveDX, Drive Genius 3, TechTool Pro 8, DiskDrill, SMARTReporter, CheckMate and possibly other disk utilities to monitor S.M.A.R.T., etc.

-Al-

On Thu, Oct 01, 2015 at 12:01 AM, casey1234 wrote:
> @alvarnell <https://github.com/alvarnell>:
> Does it also work with external USB drives? 
> What App do you use for reading the SMART data?

casey1234, 2015-10-02:
Okay. 
I use DriveDx (1.4.0) on OSX El Capitan (final now!) and it still doesn't work. I cannot read the SMART data of my external USB drive. 

Hoping for an update of the driver soon! :)

fiery-, 2015-10-03:
The SATSMARTDriver v0.8 that is signed by DriveDx developer works on El Capitan, but only after an upgrade from Yosemite (where the driver is already installed). Unfortunately, after the first system restart it no longer works. I noticed that only the installer pkg file is signed by DriveDx developer with a valid signature. The kext file is not signed and this is what the Console writes:
com.apple.kextcache[576]: SATSMARTDriver.kext has invalid signature; omitting.

Apple docs state that kext files should be signed and put in /Library/Extensions (not /System/Library/Extensions).

Also, I noticed that IOUSBAttachedSCSI is replaced by IOUSBMassStorageUASDriver in the ioreg output.

SMART Utility v3.2 states that it works on El Capitan, but included SATSMARTDriver v0.8 is not signed at all (neither the pkg nor the kext).

macuserguru, 2015-10-04:
I hope  this would be fixed soon.

casey1234, 2015-10-04:
I made a clean install of El Capitan (no update via the app store) and cannot display any SMART information of my external USB drives with DriveDx. 
We really need it fixed! :-/

alvarnell, 2015-10-04:
Then don't waste time posting here, plead your case directly to DriveDx and SMART Utility. I'm sure it's in their best interest to have it working and have enough money to make it happen.

macuserguru, 2015-10-04:
I had send an email to DriveDX support and describe them the problem too.
Please make this too, the more the better!

kasbert, 2015-10-04:
I just upgraded to El Capitan. The driver survived the install and worked just fine. Then I removed the driver and reinstalled, and the driver stopped working.

apple4ever, 2015-10-04:
I'm the developer of SMART Utility. I would be happy to work with @kasbert to provide a signed kext or other installer help to get this working in El Capitan for SMART Utility, DriveDx, and any other apps that would use.

kasbert, 2015-10-04:
@apple4ever thanks, it would be nice. I guess the build script needs updating.

The current version is somewhat work in progress. The driver will not work on unknown devices. Can be enabled with a flag, though.
The idea was to create a preferences panel and a configuration daemon. For unknown devices the driver would be enabled manually and the configuration daemon would disable the device automatically after crash. I also was thinking of the driver sending reports to my server on new or broken enclosures. In that way I could gather usb ids for next release.

casey1234, 2015-10-05:
@alvarnell @macuserguru
I emailed DriveDx two weeks ago, no reply. I don't know if they are working on a fix for the app.

Wow, this has already created a stir. I am very happy, that you are working on an update together! :-)

core-code, 2015-10-05:
i'm the developer of SMARTReporter and also thankful for this extension. i'd be happy to help in any way to keep the driver working under El Capitan, e.g. by covering the costs of the Apple Developer Membership - just let me know.

note that KEXT signing is not available to all developers anymore but must be 'specially requested'

 https://developer.apple.com/contact/kext/

i hope Apple doesn't make problems there but just grants the necessary permissions....

apple4ever, 2015-10-05:
Disappointing that Apple made kext signing a special request. I wonder what their requirements would be for that, and if if this would meet it. I would hope so. 

So @kasbert where do you want to go from here? Do you want to get an Apple Developer Membership? Or would you prefer one of us use ours? 

I think a prefpane would be nice to have. I could work on that.

alvarnell, 2015-10-07:
I heard back from DriveDx just now:

If you are using standalone (non-AppStore) edition of DriveDx - update it to v1.4.1 (menu "DriveDx > Check For Updates...").
If you are using AppStore edition of DriveDx - you should download updated driver from our website - http://binaryfruit.com/download/mac/satsmartdriver/SATSMARTDriver-0.8.zip.

Please learn more here - http://binaryfruit.com/drivedx/kb#external-drives-support-osx-el-capitan

casey1234, 2015-10-07:
I got the same Email. They obviously don't know about the problem and also didn't test El Capitan so far. What the heck are they doing?? Actually they should already work on a solution. -.-

alvarnell, 2015-10-07:
DriveDx v1.4.1 with their version of SAT-SMART-Driver 0.8 is working perfectly for me with USB drives now.

The only thing I saw wrong was their instructions in the Knowledge Base for removing the old version. You must disable SIP from the Recover HD before those commands will work.

-Al-

On Wed, Oct 07, 2015 at 01:18 AM, casey1234 wrote:
> I got the same Email. They obviously don't know about the problem and also didn't test El Capitan so far. What the heck are they doing?? Actually they should already work on a solution.

casey1234, 2015-10-07:
And what, if I don't want to disable SIP? Then it doesn't work, right? :-/

alvarnell, 2015-10-07:
I did not have to disable SIP.

casey1234, 2015-10-07:
So you just installed DriveDx 1.4.1 and the driver 0.8 under El Capitan and it worked?

alvarnell, 2015-10-07:
Exactly.

casey1234, 2015-10-07:
But you wrote you had to disable SIP for uninstalling, right? For uninstalling DriveDx or the SMART driver?

alvarnell, 2015-10-07:
I previously removed the old one some time ago and am not certain how I did that, so I edited out that part of what I wrote. It no longer says that on GitHub. It would appear that /System/Library/Extensions/ is not protected by SIP.  I suppose that's because they can control it using the .kext signature program.

macuserguru, 2015-10-07:
How I could look if SIP is disabled/enabled?

macuserguru, 2015-10-07:
is it that?
(my Mac is in German language)
Systeminformationen/Systemeinstellungen/Sicherheit:
  Version:	14,0
  Unterstützt von:	Apple
  Sichtbar:	Ja
  ID:	com.apple.preference.security
  Ort:	/System/Library/PreferencePanes/Security.prefPane
  64-Bit (Intel):	Ja

SaltyAir, 2015-10-07:
@macuserguru https://developer.apple.com/library/prerelease/mac/documentation/Security/Conceptual/System_Integrity_Protection_Guide/ConfiguringSystemIntegrityProtection/ConfiguringSystemIntegrityProtection.html

core-code, 2015-10-07:
this is no troubleshooting forum for basic Mac OS X usage. 

please stick to actual discussion on fixing this bug.

macuserguru, 2015-10-07:
@core-code 
sorry - its an extraordinaire ;-)

If I understand right the binary fruit DriveDV version runs without disabled SIP?

Momentary I upgrade all my MacPorts programs, than comes the first TM backup after El Capitan upgrade, then I will test the binary fruit DriveDV version of the SAT-SMART-Driver 0.8 for El Capitan.

casey1234, 2015-10-07:
Holy moly!! I can confirm what alvarnell describes! It works again!! 

Instruction:
1. Uninstall old DriveDx Version
2. Uninstall old driver using these commands (from the DriveDx website):
sudo rm -r /System/Library/Extensions/SATSMARTDriver.kext
sudo rm -r /System/Library/Extensions/SATSMARTLib.plugin
3. Reboot
4. Install latest DriveDx Version 1.4.1
5. Install the new updated driver (for OS X El Capitan 10.11) from the DriveDx website:
http://binaryfruit.com/drivedx/usb-drive-support
6. Reboot
7. Enjoy the SMART data from your external USB drives! :-)

macuserguru, 2015-10-07:
@casey1234
is SIP enabled after?
What shown after?
$ csrutil status

I use DriveDx 1.4.0 (Build 485) from AppStore - comes an (licensed) update?

casey1234, 2015-10-07:
Hi!

I didn't do anything with SIP, it is still enabled!

I also tested it with the 1.4.0 Version, it also worked! I think the important thing is the brandnew driver from their website - here is the download link (for 10.11 El Capitan only):
http://binaryfruit.com/download/mac/satsmartdriver/SATSMARTDriver-0.8.zip

Best wishes
Casey

alvarnell, 2015-10-07:
The 1.4.1 version contains the same driver as the one on their web site, so there is no need to download it, just install it from within DriveDx.

And yes, SIP is not in play here. It remains in whatever state you currently have it.

macuserguru, 2015-10-08:
The sign driver works well now 'driver for OS X El Capitan 10.11+'
http://binaryfruit.com/drivedx/usb-drive-support/ 
I looked inside the kext packet - its included a new sign file.
After a reboot:
$ csrutil status
System Integrity Protection status: enabled.

Well done - thanks!

apple4ever, 2015-10-08:
Good to hear there is something, but I would still prefer a solution here.

rivera-ernesto, 2015-12-30:
Also because this solution is for 0.8 when we already have 0.10 released.

macuserguru, 2015-12-30:
gives another code singed version as the 0.8 from 
http://binaryfruit.com/drivedx/usb-drive-support/ 
that works an El Capitan?

PS: Binaryfruit offers an update from 0.8 too

[1 more reply]