kasbert/OS-X-SAT-SMART-Driver #41: System Profiler blocked in 10.11.3

noyannus, 2016-02-15:
The presence of OS-X-SAT-SMART-Driver makes system profler stop with an error.

When it is installed, the terminal command 
`system_profiler SPUSBDataType`
returns
...`Failed to parse plist output from '/usr/sbin/system_profiler -nospawn -xml SPUSBDataType -detailLevel full', error: Error Domain=NSCocoaErrorDomain Code=3840 "Unexpected character S at line 1" UserInfo={NSDebugDescription=Unexpected character S at line 1, kCFPropertyListOldStyleParsingError=Error Domain=NSCocoaErrorDomain Code=3840 "Unexpected ';' or '=' after key at line 1" UserInfo={NSDebugDescription=Unexpected ';' or '=' after key at line 1}}`

After removing OS-X-SAT-SMART-Driver and a reboot, system profiler runs fine.

Observed on Mac OS version 10.11.3 (15D21

lalicatala, 2016-02-16:
Running 10.11.3 and smart driver 0.10,with an SD card connected but no external hard drive connected,  and using that same terminal command, I get this:
USB:
    USB 2.0 Bus:

      Host Controller Driver: AppleUSBEHCIPCI
      PCI Device ID: 0x1c26 
      PCI Revision ID: 0x0005 
      PCI Vendor ID: 0x8086 

        Hub:

          Product ID: 0x2513
          Vendor ID: 0x0424  (SMSC)
          Version: b.b3
          Speed: Up to 480 Mb/sec
          Location ID: 0xfd100000 / 1
          Current Available (mA): 1000
          Current Required (mA): 2
          Extra Operating Current (mA): 0
          Built-In: Yes

            IR Receiver:

              Product ID: 0x8242
              Vendor ID: 0x05ac  (Apple Inc.)
              Version: 0.16
              Speed: Up to 1.5 Mb/sec
              Manufacturer: Apple Computer, Inc.
              Location ID: 0xfd110000 / 2
              Current Available (mA): 1000
              Current Required (mA): 100
              Extra Operating Current (mA): 0
              Built-In: Yes

    USB 2.0 Bus:

      Host Controller Driver: AppleUSBEHCIPCI
      PCI Device ID: 0x1c2d 
      PCI Revision ID: 0x0005 
      PCI Vendor ID: 0x8086 

        FaceTime HD Camera (Built-in):

          Product ID: 0x8509
          Vendor ID: 0x05ac  (Apple Inc.)
          Version: 5.16
          Serial Number: DJHBC704B3DH7HL0
          Speed: Up to 480 Mb/sec
          Manufacturer: Apple Inc.
          Location ID: 0xfa200000 / 2
          Current Available (mA): 1000
          Current Required (mA): 500
          Extra Operating Current (mA): 0
          Built-In: Yes

        Hub:

          Product ID: 0x2513
          Vendor ID: 0x0424  (SMSC)
          Version: b.b3
          Speed: Up to 480 Mb/sec
          Location ID: 0xfa100000 / 1
          Current Available (mA): 1000
          Current Required (mA): 2
          Extra Operating Current (mA): 0
          Built-In: Yes

            Apple Internal Keyboard / Trackpad:

              Product ID: 0x0252
              Vendor ID: 0x05ac  (Apple Inc.)
              Version: 1.18
              Speed: Up to 12 Mb/sec
              Manufacturer: Apple Inc.
              Location ID: 0xfa120000 / 4
              Current Available (mA): 1000
              Current Required (mA): 40
              Extra Operating Current (mA): 0
              Built-In: Yes

            BRCM2070 Hub:

              Product ID: 0x4500
              Vendor ID: 0x0a5c  (Broadcom Corp.)
              Version: 1.00
              Speed: Up to 12 Mb/sec
              Manufacturer: Apple Inc.
              Location ID: 0xfa110000 / 3
              Current Available (mA): 1000
              Current Required (mA): 94
              Extra Operating Current (mA): 0
              Built-In: Yes

                Bluetooth USB Host Controller:

                  Product ID: 0x821a
                  Vendor ID: 0x05ac  (Apple Inc.)
                  Version: 0.42
                  Speed: Up to 12 Mb/sec
                  Manufacturer: Apple Inc.
                  Location ID: 0xfa113000 / 6
                  Current Available (mA): 1000
                  Current Required (mA): 0
                  Extra Operating Current (mA): 0
                  Built-In: Yes

lalicatala, 2016-02-16:
Sorry, I also meant to say that when I do connect an external hard drive, SMARTReporter does report that the drive is SMART enabled.

macuserguru, 2016-02-16:
How do you get running smart driver 0.10 under 10.11.3? 
The kext must be signed with an Apple kext developer key.
I use the signed SMART driver coming with DriveDX

`----from-system_profiler---

SATSMARTDriver:

      Version: 0.8.1
      Last Modified: 18.01.14, 21:56
      Bundle ID: org.dungeon.driver.SATSMARTDriver
      Loaded: Yes
      Obtained from: Identified Developer
      Kind: Intel
      Architectures: x86_64
      64-Bit (Intel): Yes
      Location: /System/Library/Extensions/SATSMARTDriver.kext
      Kext Version: 0.8.1
      Load Address: 18446743521854000000
      Loadable: Yes
      Dependencies: Satisfied
      Signed by: Developer ID Application: Kirill Luzanov (4ZNF85T75D), Developer ID Certification Authority, Apple Root CA
      
----from-EtreCheck-----

     /System/Library/Extensions
    [loaded]    org.dungeon.driver.SATSMARTDriver (0.8.1 - SDK 10.6) [Support]
`

noyannus, 2016-02-16:
I have System Integrity Protection disabled permanently so the install was allowed. Also, my hardware is HP (hackintoshing is not illegal here), but as the driver has worked in earlier OS versions, this is unlikely to be the cause.

macuserguru, 2016-02-16:
For system_profiler SPUSBDataType I get this answer below, but all datas for the USB drives are shown.

`$ system_profiler SPUSBDataType
2016-02-16 09:24:18.903 system_profiler[1669:352463] SPUSBDevice: IOServiceGetMatchingService did not return anything for location 0x04500000
2016-02-16 09:24:18.904 system_profiler[1669:352463] SPUSBDevice: IOServiceGetMatchingService did not return anything for location 0x04600000
2016-02-16 09:24:18.906 system_profiler[1669:352463] SPUSBDevice: IOServiceGetMatchingService did not return anything for location 0x06110000
2016-02-16 09:24:18.907 system_profiler[1669:352463] SPUSBDevice: IOServiceGetMatchingService did not return anything for location 0x06100000
USB:

    USB 2.0 Bus:

      Host Controller Driver: AppleUSBEHCIPCI
      PCI Device ID: 0x0aa6 
      PCI Revision ID: 0x00b1 
      PCI Vendor ID: 0x10de 
........
`

macuserguru, 2016-02-16:
@noyannus 
OK ... I use an older MBP 15" late 2008 with 10.11.3
I have an DriveDX AppStore license too.

I hope the latest SMART kext driver get an Apple kext developer sign soon.

macuserguru, 2016-02-16:
@noyannus 
if I know right - hackintosh require to disable SIP for the hackintosh kext drivers?

alvarnell, 2016-02-16:
It’s not necessary to disable all of SIP in order to load unsigned kernel extensions, just use enable —without kext.

Based on emerging threats, keeping as much of System Integrity Protection enabled will be increasingly important.  According to the Mackintosh instructions I have read on the subject, they “...recommend that SIP be disabled from the beginning of the installation through post-installation process. After everything is set, and the user is successfully booting, SIP can be re-enabled."

Perhaps you have special requirements for v0.10, but all of my needs are currently met with v0.81.

-Al-

On Tue, Feb 16, 2016 at 12:26 AM, noyannus wrote:
> I have System Integrity Protection disabled permanently so the install was allowed. Also, my hardware is HP (hackintoshing is not illegal here), but as the driver has worked in earlier OS versions, this is unlikely to be the cause,
>

macuserguru, 2016-02-16:
The 
/System/Library/Extensions/SATSMARTDriver.kext/Contents/MacOS/SATSMARTDriver (version 0.8.1)
and 
/System/Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib (version 1.0)
 are from 9. Dec. 2015

lalicatala, 2016-02-16:
I connected an external WD My Passport 2 TB USB3 to USB2 to my late 2011 MBP running 10.11.3. 

I ran Etrecheck and got this:
/System/Library/Extensions
[loaded] fi.dungeon.driver.SATSMARTDriver (0.10.1 - SDK 10.6 - 2016-02-10) [Support]

I ran the terminal command and got this:
    USB 2.0 Bus:

      Capacity: 2 TB (2,000,365,289,472 bytes)
      Removable Media: Yes
      BSD Name: disk4
      Host Controller Driver: AppleUSBEHCIPCI
      Partition Map Type: GPT (GUID Partition Table)
      PCI Device ID: 0x1c26 
      PCI Revision ID: 0x0005 
      PCI Vendor ID: 0x8086 
      S.M.A.R.T. status: Verified
      Volumes:
        EFI:
          Capacity: 209.7 MB (209,715,200 bytes)
          BSD Name: disk4s1
          Content: EFI
          Volume UUID: 0E239BC6-F960-3107-89CF-1C97F78BB46B
        disk4s2:
          Capacity: 2 TB (1,999,505,530,880 bytes)
          BSD Name: disk4s2
          Content: Apple_CoreStorage
        Recovery HD:
          Capacity: 650 MB (650,002,432 bytes)
          BSD Name: disk4s3
          Content: Apple_Boot
          Volume UUID: 07F0D4F7-8F16-385F-830B-6DB3A40DEC27

        Hub:

          Product ID: 0x2513
          Vendor ID: 0x0424  (SMSC)
          Version: b.b3
          Speed: Up to 480 Mb/sec
          Location ID: 0xfd100000 / 1
          Current Available (mA): 1000
          Current Required (mA): 2
          Extra Operating Current (mA): 0
          Capacity: 2 TB (2,000,365,289,472 bytes)
          Removable Media: Yes
          Detachable Drive: Yes
          BSD Name: disk4
          Built-In: Yes
          Partition Map Type: GPT (GUID Partition Table)
          S.M.A.R.T. status: Verified
          Volumes:
            EFI:
              Capacity: 209.7 MB (209,715,200 bytes)
              BSD Name: disk4s1
              Content: EFI
              Volume UUID: 0E239BC6-F960-3107-89CF-1C97F78BB46B
            disk4s2:
              Capacity: 2 TB (1,999,505,530,880 bytes)
              BSD Name: disk4s2
              Content: Apple_CoreStorage
            Recovery HD:
              Capacity: 650 MB (650,002,432 bytes)
              BSD Name: disk4s3
              Content: Apple_Boot
              Volume UUID: 07F0D4F7-8F16-385F-830B-6DB3A40DEC27

            My Passport 0820:

              Product ID: 0x0820
[... 100 lines cut ...]
              Extra Operating Current (mA): 0
              Built-In: Yes

                Bluetooth USB Host Controller:

                  Product ID: 0x821a
                  Vendor ID: 0x05ac  (Apple Inc.)
                  Version: 0.42
                  Speed: Up to 12 Mb/sec
                  Manufacturer: Apple Inc.
                  Location ID: 0xfa113000 / 6
                  Current Available (mA): 1000
                  Current Required (mA): 0
                  Extra Operating Current (mA): 0
                  Built-In: Yes

So, for the person that asked me how I got 0.10 to run, I installed out with the installer and……… did nothing much else. 

I did ensure that any remnant of and previous version of SATSMART was removed from the system prior to installing 0.10

From SmartReporter version 3.1.7, it reports:
S.M.A.R.T. status check & self-test found 2 disk(s) O.K. and 0 disk(s) not S.M.A.R.T. capable.
S.M.A.R.T.-checked disks: LeeZeMBP (JR10006P18DKGF) Passport_2GB (WD-WXR1E63HRRJ)

Sorry, cannot be of more help

lalicatala, 2016-02-16:
Sorry I made some typing errors above but the gist I hope is clear. As for the comment that I installed 0.10 because I might have had a special requirement  need, I do not. I tried the latest version that I could find and installed it, and then checking to see if it did the trick. It did so I stopped. I did not try 0.11 as the file is listed with the word "debug." I thought that meant all the the "bugs" have yet to be worked out sooooo I avoided that version.

macuserguru, 2016-02-16:
@lalicatala 
did you disable SIP?
I got the signed SMART driver from here
http://binaryfruit.com/drivedx/usb-drive-support

alvarnell, 2016-02-16:
It’s my understanding that all versions since 0.8 have been for special needs/requests.  Both SMART Utility and DriveDx are staying with 0.8 and the signed version 0.81.

-Al-

On Feb 16, 2016, at 2:28 AM, lalicatala <notifications@github.com> wrote:

> Sorry I made some typing errors above but the gist I hope is clear. As for the comment that I installed 0.10 because I might have had a special requirement need, I do not. I tried the latest version [...]

lalicatala, 2016-02-16:
Al, I guess I must have missed that. However, if something should happen to version 0.10, will then try 0.8.

macuserguru, 2016-02-16:
If I know right Kirill Luzanov has updated the database of provided drives in 0.8.1

`Version: 0.8.1
  Last Modified: 18.01.14, 21:56
  Bundle ID: org.dungeon.driver.SATSMARTDriver
 ......
  Signed by: Developer ID Application: Kirill Luzanov `

noyannus, 2016-02-17:
Did some testing and found the issue to be in 0.11debug, but not in 0.10release which runs fine. 

More detailed: Issue only appeared with disks plugged into USB port, also via a hub. After unplugging the disks, system profiler ran OK. 

Other observation: removing 0.11 required sudo rm -rf (not -r) to override lacking permissions. 

@alvarnell 
I agree about threats etc. But I rely also on TotalFinder (that died by SIP because it uses code injection) and have yet to switch to a replacement (there a many, just need to find some time to test and select what matches best my working needs and habits), so I decided to accept security at Yosemity level for some time longer.

Thanks for all your input!

alvarnell, 2016-02-17:
Actually, 0.8.1 is just a signed version of 0.8, but he also signed a v0.10.1 which does have contain the additional drives.

-Al-

On Feb 16, 2016, at 4:29 AM, macuserguru <notifications@github.com> wrote:

> If I know right Kirill Luzanov has updated the database of provided drives in 0.8.1
> 
> Version: 0.8.1
> Last Modified: 18.01.14, 21:56
> Bundle ID: org.dungeon.driver.SATSMARTDriver
> ......
> Signed by: Developer ID Application: Kirill Luzanov

closed 2016-02-18 by noyannus