kasbert/OS-X-SAT-SMART-Driver #42: el Capitan 10.11.6 thunderbolt to firewire issue

maz4200, 2016-09-02:
After this update computer will not boot with firewire drive attached or read usb3 external drives.
attaching after startup drives do not appear on desktop or at command line.
Issue solved by uninstalling kexts.

kasbert, 2016-09-05:
I'm sorry to hear that. I have also a thunderbolt to firewire adapter, but I have no issues with it. 
What is the brand and model of your external drive ?

Eakster, 2017-03-30:
I also have the same issue with SATSMARTDriver-0.10 but SATSMARTDriver-0.8 seems to boot but then all drive access is slow and freezes.

I have a iMac13,2 (Early 2008 27") with MacOSX 10.12.3 and DriveDX 1.5.1.530
Model Family                    : Western Digital Green 3TB
Model                                : WDC WD30EZRX-00MMMB0
Firmware Version              : 80.00A80
and 
Model Family                         : Western Digital Caviar Green (AF) 2TB
Model                                     : WDC WD20EARS-00S8B1
Firmware Version                  : 80.00A80

2 other Old firewire to Thunderbolt 2 bay enclosures that won't get SAT info.