kasbert/OS-X-SAT-SMART-Driver #43: SMART driver under macOS Sierra

macuserguru, 2016-09-21:
Is any version of OS-X-SAT-SMART-Driver macOS Sierra ready?

alvarnell, 2016-09-21:
Why do you think it doesn't work? I don't recall the last time an upgrade was required for a new OS.

Sent from Janet's iPad

-Al-

On Sep 21, 2016, at 9:24 AM, macuserguru wrote:
> Is any version of OS-X-SAT-SMART-Driver macOS Sierra ready?
>

alvarnell, 2016-09-22:
Sorry, I should have mentioned that since El Capitan you have to either disable the SIP requirement that .kext files be signed or use the signed one provided by DriveDx 
<https://binaryfruit.com/blog/we-published-sat-smart-driver-v0-10-for-os-x-10-10-yosemite-and-newer <https://binaryfruit.com/blog/we-published-sat-smart-driver-v0-10-for-os-x-10-10-yosemite-and-newer>>, but I’m sure you remember that discussion from last year.

-Al-

On Wed, Sep 21, 2016 at 04:46 PM, Al Varnell wrote:
> Why do you think it doesn't work? I don't recall the last time an upgrade was required for a new OS.
> 
> Sent from Janet's iPad
> 
> -Al-
> 
> On Sep 21, 2016, at 9:24 AM, macuserguru wrote:
>> Is any version of OS-X-SAT-SMART-Driver macOS Sierra ready?

elvey, 2016-10-02:
Hmm.  I wonder if I hosed my system by installing version .8 and version .10  
Just saw the warning at binaryfruit not do do that, and think I may well have; my system became unbootable and I had to reinstall the OS.

kasbert, 2016-10-02:
Usually it should boot, if you remove all external drives.

yifuhood, 2016-11-04:
I lost all my data on my sandisk extreme 500 SSD USB 3.0. I wanted to check the SSDs heath using DriveDx, and then everything just froze , I had to force eject , now  photos on the disk are all gone.

macuserguru, 2016-11-04:
@yifuhood 
Bad news! 
Off topic - its recommend to use two backups, an old admin adage: 'one backup are non-backup'
I use two USB HDs for my TM backup alternately.

yifuhood, 2016-11-04:
@macuserguru thanks. I have most the photos backed up, some new ones are not.  now I am afraid to use DriveDX , and the driver for smart  USB. so i can not tell if my sandisk extreme portable SSD has any damages after that incident .  not only the data are gone , looks like the drive was repartitioned.  DriveDx recommend to install the V0.8, so i did. should i install the V1.0 and try again?