kasbert/OS-X-SAT-SMART-Driver #44: Compiler fail on SDK macosx10.12

kingeaglewang, 2016-11-25:
Sorry, I new to the driver.

I use mac OS X with 10.12 and Xcode version is 8.1

I have change the SDK from 10.6 to 10.12 in this file SATSMARTDriver.xcodeproj/project.pbxproj

After this change, I make again.

But got this error "the clang compiler does not support 'fapple-kext' for C++ on Darwin/i386"

How can I get rid of this error ?

Thanks for helping me.

alvarnell, 2016-11-25:
Unless you have an Apple Developer ID to sign the .kext you are wasting your time.

Simply download one of the signed drivers from [DriveDX](https://binaryfruit.com/drivedx/usb-drive-support).

kingeaglewang, 2016-12-01:
I have a Apple Developer ID now.

How can I do to build this driver successfully ?

Thanks.

alvarnell, 2016-12-01:
If you don't mind my asking, what's wrong with the ones available from DriveDx?

kingeaglewang, 2016-12-01:
Because the device I use is not in the support list, I would like to modify the code.

Sorry for this crazy thinking.

kasbert, 2016-12-01:
XCode will upgrade the project for you. Anyway I updated the build files for XCode 8.1. I didn't test it though. 
You may still need to run your Mac in developer mode to run it.

kingeaglewang, 2016-12-01:
Thanks for helping solve this build error.

Although I have tried my best to read what you modify with previous version, I still don't know why this error is gone.

It is so amazing.
Thanks again.

kingeaglewang, 2016-12-01:
Sorry for asking about the makefile.

The previous version for copying SATSMARTLib.plugin is to /System/Library/Extensions, but this version is to /Library/Extensions.

Does this affect for installing kext ?

Thanks

kasbert, 2016-12-01:
/Library/Extensions is the correct place for 3rd party kexts, the location was wrong earlier. Now, I don't know if it is the correct place for plugins, but I suppose so. Even better place for the plugin would be inside the kext, but I couldn't get it to work.

kingeaglewang, 2016-12-06:
Compile is successful, but when I want to install the kernel extension, there is an error.

After this command: sudo kextutil -t /tmp/SATSMARTDriver.kext
The following is shown in the Terminal.

Notice: SATSMARTDriver.kext has debug properties set.
Diagnostics for SATSMARTDriver.kext:
Authentication Failures: 
    File owner/permissions are incorrect (must be root:wheel, nonwritable by group/other): 
        SATSMARTDriver.kext
        Contents
        Info.plist
        MacOS
        SATSMARTDriver
        Resources
        English.lproj
        InfoPlist.strings

Warnings: 
    The booter does not recognize symbolic links; confirm these files/directories aren't needed for startup: 
        SATSMARTDriver.kext

Code Signing Failure: not code signed

Any suggestion for this error ?
Thanks

kasbert, 2016-12-06:
Try disabling system integrity protection.
https://developer.apple.com/library/content/documentation/Security/Conceptual/System_Integrity_Protection_Guide/ConfiguringSystemIntegrityProtection/ConfiguringSystemIntegrityProtection.html#//apple_ref/doc/uid/TP40016462-CH5-SW1

Check those permissions, too.

kingeaglewang, 2016-12-07:
I have already disabled SIP and the permission is also root:wheel.
(drwxr-xr-x  3 root  wheel  102 12  7 09:19 SATSMARTDriver.kext)

But I still have the above error.

After I modify OSBundleLibraries in Info.plist, the installation will be successful.
I modify the library version according to this command: kextlibs -xml SATSMARTDriver.kext

Now, after this command: sudo kextutil -t /tmp/SATSMARTDriver.kext
The following information is shown in Terminal.

Notice: /tmp/SATSMARTDriver.kext has debug properties set.
Diagnostics for /tmp/SATSMARTDriver.kext:
Code Signing Failure: not code signed
kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/tmp/SATSMARTDriver.kext"
kext signature failure override allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/tmp/SATSMARTDriver.kext"

I use kextstat to check, and fi.dungeon.driver.SATSMARTDriver is there.
But the SMART status of my external disk is still "Not Support".

Any suggestion for this error ?
Thanks

alvarnell, 2016-12-07:
> Try disabling system integrity protection.

Note that you can disable only the KEXT portion of SIP while keeping the rest in force with this command from the Recovery HD:

`csrutil enable --without kext`

kingeaglewang, 2016-12-07:
Thanks alvarnell.
After this command `csrutil enable --without kext`, the `_csrutil status_ ` will be like following.

system Integrity Protection status: enabled (Custom Configuration).

Configuration:
        Apple Internal: disabled
        Kext Signing: disabled
        Filesystem Protections: enabled
        Debugging Restrictions: enabled
        DTrace Restrictions: enabled
        NVRAM Protections: enabled
        BaseSystem Verification: enabled

This is an unsupported configuration, likely to break in the future and leave your machine in an unknown state.

But after this command: `sudo kextutil -t /tmp/SATSMARTDriver.kext`
The following is shown in the Terminal.

Notice: SATSMARTDriver.kext has debug properties set.
Diagnostics for SATSMARTDriver.kext:
Authentication Failures:
File owner/permissions are incorrect (must be root:wheel, nonwritable by group/other):
SATSMARTDriver.kext
Contents
Info.plist
MacOS
SATSMARTDriver
Resources
English.lproj
InfoPlist.strings

Warnings:
The booter does not recognize symbolic links; confirm these files/directories aren't needed for startup:
SATSMARTDriver.kext

Code Signing Failure: not code signed

Any suggestion for this error ?
Thanks

kingeaglewang, 2016-12-07:
Sorry, I find what is wrong.
In my /tmp/SATSMARTDriver.kext/ there is one more SATSMARTDriver.kext.
It should be only a Contents.
After I delete it SATSMARTDriver.kext in the /tmp/SATSMARTDriver.kext/, the install will be successful.

When I `./smart_sample -a`, it prints "No ATA devices found", but I really connect a USB device to my Mac.

What's wrong is this situation?
Thanks

kasbert, 2016-12-07:
Try looking kernel messages and use ioreg-command.

kingeaglewang, 2016-12-08:
I read the /var/log/system.log, but there is no message pop out when I install SATSMARTDriver.kext.

`ioreg | grep SATSMARTDriver` nothing is returned.

However, `kextstat | grep SATSMARTDriver` returns the following.
 174    0 0xffffff7f839e1000 0x16000    0x16000    fi.dungeon.driver.SATSMARTDriver (0.11) 1396939A-AC41-39FA-83D4-7F051565918E <173 65 16 5 4 3>

Thanks

kasbert, 2016-12-09:
Try also ioreg -l and try to find the device to see what driver it is using instead.
If you use the debug version of the driver, you should see probing results.

kingeaglewang, 2016-12-12:
Sorry for the late reply.

After the long research, I found two things.

1. Mac OSX 10.12.1 does not store the kennel log in the system.log that's why I found nothing in the system.log.
According to this web site: [https://eclecticlight.co/2016/09/28/known-bugs-in-macos-sierra-10-12-an-incomplete-summary/](url).

A person named 邱韜瑋 reply as following:
Because the the log doesn’t store in the system.log anymore. You need to use log system to access the output from IOLog. Use `sudo log collect` to generate system_log.logarchive, and open this file with Sierra console app. It’s really works for me.

By using his method, I can see the IOLog in system_log.logarchive.

2. I use `make install`, but the SMART of the disk is still "Not Support".
After I unmount and mount the device manually, the SMART of the disk becomes "Verified".
And the command `ioreg | grep fi_dungeon_driver_IOSATDriver` will return something.

Thanks for helping me.

kasbert, 2016-12-12:
Would you tell the neded changes (or add a github pull request) ?

kingeaglewang, 2016-12-13:
It seems nothing has to be changed.
Code is fine.
Probably, the reason is macOSX 10.12 differs from previous version.

Thanks.