kasbert/OS-X-SAT-SMART-Driver #46: OWC Mercury External Enclosures compatible?

JDW1, 2017-05-16:
OWC Mercury Elite Pro external drives are rather popular, so it is somewhat surprising to not see them listed at all, not even in the "incompatible" section:

https://eshop.macsales.com/shop/firewire/1394/USB/EliteAL/eSATA_FW800_FW400_USB

kasbert, 2017-05-16:
It often works, even if it not in the lists.

I suppose you could send an email to smartmontools list .
https://www.smartmontools.org/wiki/FAQ#Howtocreateabugreport

JDW1, 2017-05-17:
I was able to see "verified" on my MacOS Sierra 15" 2015 MacBook Pro after a restart.  I had previously tried Logout/Login but that did not produce "Verified" in Disk Utility.  A restart is required.  Anyway, my external drive enclosures are still not listed on the compatibility chart, so I tried using the Terminal to run the SHELL line you propose, but what I see is as follows:

```
JDW$ ioreg -r -w 0 -c fi_dungeon_driver_IOSATDriver  | egrep 'Enclosure|PassThroughMode|Capable'|grep -v
usage: grep [-abcDEFGHhIiJLlmnOoqRSsUVvwxZ] [-A num] [-B num] [-C[num]]
	[-e pattern] [-f file] [--binary-files=value] [--color=when]
	[--context[=num]] [--directories=action] [--label] [--line-buffered]
	[--null] [pattern] [file ...]
```

So that line is failing and it's simply telling me about "grep" usage.  Therefore, what is the properly line to paste into the MacOS Terminal that will yield the 3 identifiers?

"Enclosure Name" 
"SATSMARTCapable"
"PassThroughMode"

Thanks.

kasbert, 2017-05-17:
The complete command is

ioreg -r -w 0 -c fi_dungeon_driver_IOSATDriver  | egrep 'Enclosure|PassThroughMode|Capable'|grep -v Identifiers

JDW1, 2017-05-18:
Thank you, but when I copied and pasted the following line in the MacOS Sierra Terminal and then hit RETURN, all I got was another $ prompt:

```
JDW$ ioreg -r -w 0 -c fi_dungeon_driver_IOSATDriver | egrep 'Enclosure|PassThroughMode|Capable'|grep -v Identifiers
```

In other words, after trying the above line in the Terminal, I do not get the following information:

"Enclosure Name"
"SATSMARTCapable"
"PassThroughMode"

Please advise.

xz4gb8, 2017-05-18:
That looks like you do not have the OS-X-SAT-SMART-Driver kexts installed. It can also happen if no candidate drives are detected and connected.

Locally, I see
$ ioreg -r -w 0 -c fi_dungeon_driver_IOSATDriver | egrep 'Enclosure|PassThroughMode|Capable'|grep -v Identifiers
  |   "Enclosure Name" = "Unknown 0424:4060"
  |   "PassThroughMode" = "auto"
  |   "PassThroughMode" = "sat16"
  |   "SATSMARTCapable" = Yes
  |   "Enclosure Name" = "Macally G-S350SU"
  |   "PassThroughMode" = "sat16"
  |   "SATSMARTCapable" = Yes
  |   "Enclosure Name" = "Macally G-S350SU"

"Enclosure Name" = "Macally G-S350SU" appears to correspond to my Mercury Pro enclosures.

JDW1, 2017-05-18:
I assure you, I do have the driver installed and the external drive is mounted.  It shows Verified in Disk Utility:

https://cl.ly/kWK8/Image%202017-05-18%20at%209.50.55%20AM.png

Right now I am at the office so I have a different external drive attached.  (My OWC Mercury externals are at home and I will check those later.)  But the line I copied-and-pasted into the Terminal should work regardless of the make of the external drive, so long as the driver is installed and the drive is mounted, which in my case, it is, as you can see in that screenshot.  The drive at my office is a Time Machine drive, if that matters.  

For good measure, I just now relaunched the Terminal, pasted in the aforementioned line (with no mistakes, nothing missing), and when I hit RETURN, I get a new $ prompt.  That's it.

xz4gb8, 2017-05-18:
You might want to look through the output of

ioreg -r -w 0 -c fi_dungeon_driver_IOSATDriver

to see what you can see

JDW1, 2017-05-18:
But there is no "output"!

When I copy-and-paste the following line into the MacOS Sierra Terminal, I get nothing but a new $ prompt!
```
ioreg -r -w 0 -c fi_dungeon_driver_IOSATDriver
```

Please advise!

xz4gb8, 2017-05-18:
The lack of output for the listed command indicates that no members of class "fi_dungeon_driver_IOSATDriver" are found active on your machine. This indicates that no version of OS-X-SAT-SMART-Driver has been installed or an unsigned version has been installed and macOS refuses to load the kexts.

I suggest using EtreCheck (https://etrecheck.com) to determine the facts.  Look for lines like
     /System/Library/Extensions
    [loaded]    fi.dungeon.driver.SATSMARTDriver (0.10 - SDK 10.9) [Lookup]

JDW1, 2017-05-18:
Wrong again.  Here's proof via ETRECHECK:

**/System/Library/Extensions**
[loaded]    org.dungeon.driver.SATSMARTDriver (0.8 - SDK 10.6) [Lookup]

But we already knew that even before I ran ETRECHECK because if the driver had no been loaded my external drive status would not have been reported as "Verified" in Disk Utility.

Please advise.

JDW1, 2017-05-18:
I'm now at home.  Tried all the aforementioned lines in the Terminal on this Mac too, but the result is the same.  When I hit RETURN, I get a new $ prompt and that's it.

We've already established the driver is installed on both my machines.  I think there can be no doubt about that.  So why then is the Terminal command not working at all?  I even tried using SUDO before the command, but all it did was ask for my PSW and then gave me a new $ prompt.

Have I found a bug here?

Please advise.

JDW1, 2017-05-18:
Silence usually means either "I'm on vacation" or "I haven't the faintest idea," I will assume the later.  This would appear to be a bug.  I am more than happy to assist in the investigation of this bug.  Just let me know what to do and I will proceed.

At the end of the day though, it doesn't matter too much to me.  I use the driver and it works.  I merely wanted to glean the data YOU need about my OWC external drive enclosure so that data could be added to the Compatibility Chart.  But since **xz4gb8** admitted to having OWC drives and admitted to being able to glean the pertinent data from them, I would suggest that he submit that data so it can be added to the Compatibility Chart.

Thanks.

alvarnell, 2017-05-19:
I have been under the impression for some time now that further development here has stopped and that any further actions are up to Smart Utility, DriveDx and any other vendor who choses to sign and distribute it.

Please correct me if I'm wrong.

Sent from Janet's iPad

-Al-
-- 
Al Varnell
Mountain View, CA

On May 18, 2017, at 4:52 PM, James Wages wrote:
> Silence usually means either "I'm on vacation" or "I haven't the faintest idea," I will assume the later. This would appear to be a bug. I am more than happy to assist in the investigation of this [...]

JDW1, 2017-05-19:
Be that as it may, it still seems to be OUR responsibility to submit data on external drives that work, if they are not already listed on the Compatibility Chart.  And that has nothing to do with development of the driver.  So why did I chose to comment here?  Because here on Github it gives specific instructions on how to go about gleaning that data from the external drives for submission.  And I merely reported that on two separate Macs, that information is useless.  It doesn't work on either of the Macs I've tried, which is why if **xz4gb8** can kindly submit data on his external OWC drives, that would be the same as mine and this conversation is brought to a complete conclusion.

llamafilm, 2018-02-11:
@JDW1 you are not seeing it in ioreg because you're searching for the wrong class name.  Try this command: 
`ioreg -r -w 0 -c org_dungeon_driver_IOSATDriver`

JDW1, 2018-02-11:
@**llamafilm**  I opened the Terminal under MacOS High Sierra and then pasted in your suggested line and hit Return.  As was the case before, I see only a new command prompt, nothing more.  Thanks.

llamafilm, 2018-02-11:
Interesting. You did that while an external drive is connected, and the driver shows as loaded in kextstat? The class name might differ depending on which version is installed. With IOJones, search for dungeon or SAT and take a screenshot of the result. IOJones has the same info as ioreg, but in an easier-to-read format.

JDW1, 2018-02-11:
Thanks.  I connected the power adapter and tried again and it worked this time.  

I have the data saved in a *.txt file, but should I post it in this thread or somewhere else?  And should I blank out part or all the drive SN's, or does it even matter?  (I assume it doesn't.)

Thanks.

alvarnell, 2018-02-11:
The README instructions are to post it here. 

The drives model number would be more important than the serial number, although I would guess that model could be derived from serial, at least by the manufacturer. The only thing that might be sensitive about the serial number would be associated with getting warrantee service, but I would think proof of purchase would also be required and ultimately an RMA would result in discovering the actual serial number.

JDW1, 2018-02-11:
@alvarnell Thanks.  Here is the text generated by the MacOS Terminal.
[OWC_MercuryElitePro.txt](https://github.com/kasbert/OS-X-SAT-SMART-Driver/files/1714178/OWC_MercuryElitePro.txt)

mikehardy, 2018-06-07:
For any future people, just confirming that OWC Mercury enclosures seem fine along with OS X 10.13.x (High Sierra)

Backup your data before you do anything I say!

Disconnect your drives, Install the [DriveDX signed driver](https://binaryfruit.com/drivedx/usb-drive-support#SAT-SMART-Driver-v0.10-Yosemite), reboot, reconnect, and SMART pass through works (I work with 'smartctl' which is from the "smartmontools" package you may install with brew as I did)

Works fine for me.

I see this information:

  |   "PassThroughMode" = "sat12"
  |   "SATSMARTCapable" = Yes
  |   "Enclosure Name" = "Unknown fw 01d2:0000"