kasbert/OS-X-SAT-SMART-Driver #53: Whats the difference between SATSMARTDriver from binaryfruit

macuserguru, 2018-12-29:
Whats the difference between SATSMARTDriver from binaryfruit
SATSMARTDriver-0.8.pkg 09.12.2015 (dd.mm.yyyy)  MID5 = 0d539c1370dbf56777c8a3d30727e4cc
SATSMARTDriver-0.8.pkg 03.11.2017 (dd.mm.yyyy)  MID5 = f6deb6585a5a0250ad7017c01d41492d 

My on ElCapitan installed version:
/System/Library/Extensions/SATSMARTDriver.kext  ver.0.8.1   18.01.2014 (dd.mm.yyyy)
/System/Library/Extensions/SATSMARTLib.plugin   ver.1.0      18.01.2014 (dd.mm.yyyy)

Which driver version included SATSMARTDriver-0.8.pkg 03.11.2017 (dd.mm.yyyy)?
Or only are the supported USB devices different?

macuserguru, 2018-12-29:
Additional:
Is DriveDX are not longer on Apple App Store available? I don't found DriveDX or Binary Fruit in App Store.
Another guy tells me, the DriveDX  developer don't answer him.

alvarnell, 2018-12-29:
The only difference was that utility versions are codesigned (with permission granted here) to load on recent macOS releases.

No drive utilities are available from the App Store as they can't do their jobs and conform to Apple's App Store rules.