kasbert/OS-X-SAT-SMART-Driver #54: Which SAT SMART driver for Mojave?

macuserguru, 2018-12-29:
Which SAT SMART driver versions are Mojave ready? 
Are for Thunderbolt drives (external disks in TB cases) a SAT SMART driver needed too?

macuserguru, 2018-12-30:
Thanks!
The open question:
Which SAT SMART driver versions are Mojave ready?

ferebee, 2018-12-30:
I use the 0.8 signed version kindly provided by the DriveDx folks. It works fine with Mojave 10.14.2 and has worked with almost all USB bridge boards I’ve tried it with. A few rare USB bridges don’t support SMART no matter what, and only once did I find one that caused a kernel panic in SAT-SMART. (Easy solution: ditch the offending USB bridge.)

> Am 30.12.2018 um 14:15 schrieb macuserguru <notifications@github.com>:
> 
> Thanks!
> The open question:
> Which SAT SMART driver versions are Mojave ready?

abdulmannan, 2019-03-22:
On Mojave 10.14.3, DriveDx's driver V0.8 did not work. However, V0.10 worked perfectly.

macuserguru, 2019-03-23:
On ElCapitan V0.8 from DriveDX compiled and signed works perfectly.
V0.10 I don't test them.

Th3Crow, 2019-04-13:
I am not having any problems with DriveDx's 0.8 driver with Mojave 14.4

altimac, 2019-05-13:
DriveDx's v0.8 works on Mojave 10.14.4 here. First install, never installed any other version before.