kasbert/OS-X-SAT-SMART-Driver #55: What the future for this project?

macuserguru, 2018-12-29:
Will there also be further development in the future?
Adaptation to new packages and their chip sets?

macuserguru, 2018-12-30:
Is there a similar project somewhere?

alvarnell, 2018-12-30:
None that I am aware of. All the drive utility developers seem to have adopted this and turned it into their own solution.

bitnautist, 2019-08-11:
This no longer works under Catalina macOS 10.16 beta 5...if there's no further development then Mojave is the last macOS that will run this driver. :(

macuserguru, 2019-08-12:
Bad news! I hope for SMART drivers for Catalina too!
I want to buy a new Mac in the fall, until then it would be good if the problem was solved.

alvarnell, 2019-08-12:
It's evident that the developer here has abandoned the project, so I suspect the only hope is that another product developer that currently uses SAT-SMART drivers (e.g. SMART Utilities, DriveDx) will come up with a solution.

apple4ever, 2019-08-18:
All - I just tested this under Catalina Beta 5, and it worked for me. 

As the developer of SMART Utility, I will do my best to keep the driver working. I also do hope to update it and fix a few bugs.

alvarnell, 2019-08-18:
Great!
So @bitnautist, what specifically are you seeing that isn’t working for you?

bitnautist, 2019-08-18:
Catalina beta 5 says the driver is not compatible...

[image]

alvarnell, 2019-08-18:
DriveDx is going to have to solve that part. It's their installer.

apple4ever, 2019-08-19:
I've seen that with my installer too, but I have yet to be able to reproduce it. 

What Mac are you installing it on? 

And where?

bitnautist, 2019-08-19:
I’m installing on a Mac Pro 6,1 with 6 cores and 64GB of RAM.

BinaryFruit, 2019-09-16:
Hi All,
macOS Catalina 10.15 ready drivers (v0.8 and v0.10) are available on [our website](https://binaryfruit.com/drivedx/usb-drive-support) and included in DriveDx v1.8.3. 
Enjoy!