kasbert/OS-X-SAT-SMART-Driver #58: Not working on macOS Catalina 10.15.2

Smexhy, 2020-01-28:
Unfortunately I bought DriveDx software which is using this driver but apparently it’s not working anymore in the lates macOS release. Nobody is answering on support emails which is shame because I don’t really want to, really liked the program and this is the only alternative for mac. If you guys consider this software for external disk tests then don’t use it for now.

killerovsky, 2020-01-28:
> Not that it’s only “not-working” but it also makes external drives unusable on mac where the driver is installed.

Driver is working without problems on all my 3 Macs under macOS Catalina 10.15.2.

If you want to make your report a little useful for the Mac community, please add specific details about your configurations:
1) The exact Mac model and year of manufacture, and does it have a T2 chip?
2) Models of used external drives or models of external boxes or docks or hubs, etc.
3) FireWire, USB2 or USB3?
4) What driver versions did you use?
5) You say that it does not work for you on 10.15.2 - but did it work on 10.15.1 or 10.15.0?

alvarnell, 2020-01-29:
Make sure you have the latest version 1.9.1 of DriveDx. There was a change of location of the SAT-SMART driver in that update which allows it to work with Catalina.

-Al-

Smexhy, 2020-02-01:
I use it only because I bought DriveDx and yes I am using the latest version (at least of DriveDx, not sure which version they use).

I wrote multiple emails to them with logs and information about my mac and versions, however no response until today.

I just wanted you to know the background behind this.

I am using iMac late 2013 27inch, intel i7 - so without T2 chip and I don’t know if the driver was working back then I tried it just now on 10.15.2 together with drivedx when I bought it, so the version of driver they provide (not sure which is that).

It’s USB3.

This is hub I use: https://www.ugreen.com/product/UGREEN_Black_USB_to_Sata_Adapter_for_SATA_Hard_Drives-en.html

Not sure if it helps but SMART data are displayed just fine when using this hud with windows and software like HDD sentinel.

Drives I used differs from manufacturers like WD, samsung, sandisk, toshiba, hitachi...tried really everything I could to pinpoint if it’s faulty disk or not.

Smexhy, 2020-02-01:
> Make sure you have the latest version 1.9.1 of DriveDx. There was a change of location of the SAT-SMART driver in that update which allows it to work with Catalina.
> […](#)

This is the version I am using from the start

killerovsky, 2020-02-04:
@Smexhy I also post the text of our answer here in case you suddenly miss (do not receive) our email again:
 
1. We replied to you on November 26, 2019. If you didn’t receive it please check the “spam folder” in your email client. 

2. The issue you are describing is not related to DriveDx but to your specific USB-SATA adapter or/and to the third party free open source driver.
 
3. On our website it is clearly stated that driver is not our project. 
Sometimes we can help and assist user with the driver and we do this quite often, 
but we have never promised it anywhere and the priority also depends on the current load of support.
 
4. Your UGREEN adapter is not S.M.A.R.T. capable (do not translate S.M.A.R.T. diagnostic commands) not only on macOS but on Linux too:
https://www.amazon.com/gp/customer-reviews/R2N2OQNG9UNUAG/ref=cm_cr_arp_d_viewpnt?ie=UTF8&ASIN=B01H1I9X60#R2N2OQNG9UNUAG
So, it looks like adapter and/or its firmware is not fully compatible with OS’s other then Windows
 
5. If your main goal is to perform drive diagnostics on Mac - simply use another SATA-USB adapter/enclosure/dock.
 
P.S. Also, it is not fully clear why you have decided to manually "improve" the installation of the already installed driver. So, as a result, there is some possibility that you somehow incorrectly (re)installed driver and as a result, you now have outdated or corrupted kext/driver cache in macOS. When the macOS boots - it saves all currently used kexts (drivers) into a cache for faster booting next time - and if you move things around in “/Library/Extensions/” and “/System/Library/Extensions/” - you won't see those changes reflected until the caches are rebuilt.
To rebuild cache - type in the terminal following commands:
```
sudo kextcache -i /
sudo kextcache -u /
sudo kextcache -system-caches
```

killerovsky, 2020-02-04:
Every few months, dozens, if not hundreds, of new external drive models are released, so it is absolutely impossible to predict the
compatibility of all combinations of hardware configurations (external drive model <-> specific version of macOS <-> specific model
of Mac). Moreover, external drive internals (enclosure (SATA-to-USB) controller, firmware etc) could vary significantly depending on market and manufacturing date.
So, there are almost endless number of combinations as there are thousands of external drive models, dozens of Mac models and generations, many macOS versions
There are almost endless number of combinations as there are thousands of external drive models, dozens of Mac models and generations, many macOS versions.
So, there is only one reliable way to make sure - to check in real life on real equipment.

BinaryFruit, 2020-02-04:
@Smexhy 
1. We replied to you on November 29, 2019. If you didn’t receive it please check the “spam folder” in your email client.

2. The issue you are describing is not related to DriveDx but related to your specific USB-SATA adapter or/and to the third party free open source driver.
 
3. On our website it is clearly stated that driver is not our project. 
Sometimes we can help and assist user with the driver and we do this quite often, 
but we have never promised it anywhere and the priority also depends on the current load of support.

4. Your UGREEN adapter is not S.M.A.R.T. capable (do not translate S.M.A.R.T. diagnostic commands) not only on macOS but on Linux too:
https://www.amazon.com/gp/customer-reviews/R2N2OQNG9UNUAG/ref=cm_cr_arp_d_viewpnt?ie=UTF8&ASIN=B01H1I9X60#R2N2OQNG9UNUAG
So, it looks like adapter and/or its firmware is not fully compatible with OS’s other then Windows
 
5. If your main goal is to perform drive diagnostics on Mac - simply use another SATA-USB adapter/enclosure/dock.
 
P.S. Also, it is not fully clear why you have decided to manually "improve" the installation of the already installed driver. So, as a result, there is some possibility that you somehow incorrectly (re)installed driver and as a result, you now have outdated or corrupted kext/driver cache in macOS.
When the macOS boots - it saves all currently used kexts (drivers) into a cache for faster booting next time - and if you move things around in “/Library/Extensions/” and “/System/Library/Extensions/” - you won't see those changes reflected until the caches are rebuilt.
Type in the terminal following commands:
```
sudo kextcache -i /
sudo kextcache -u /
sudo kextcache -system-caches
```

BinaryFruit, 2020-02-04:
Every few months, dozens, if not hundreds, of new external drive models are released, so it is absolutely impossible to predict the
compatibility of all combinations of hardware configurations (external drive model <-> specific version of macOS <-> specific model
of Mac). Moreover, external drive internals (enclosure (SATA-to-USB) controller, firmware etc) could vary significantly depending on market and manufacturing date.
So, there are almost endless number of combinations as there are thousands of external drive models, dozens of Mac models and generations, many macOS versions
There are almost endless number of combinations as there are thousands of external drive models, dozens of Mac models and generations, many macOS versions.
So, there is only one reliable way to make sure - to check in real life on real equipment.

Smexhy, 2020-02-04:
> @Smexhy
> 
> 1. We replied to you on November 26, 2019. If you didn’t receive it please check the “spam folder” in your email client.
> 2. The issue you are describing is not related to DriveDx but related to your specific USB-SATA adapter or/and to the third party free open source driver.
> 3. On our website it is clearly stated that driver is not our project.
>    Sometimes we can help and assist user with the driver and we do this quite often,
>    but we have never promised it anywhere and the priority also depends on the current load of support.
> 4. Your UGREEN adapter is not S.M.A.R.T. capable (does not translate S.M.A.R.T. diagnostic commands) not only on macOS but on Linux too:
>    https://www.amazon.com/gp/customer-reviews/R2N2OQNG9UNUAG/ref=cm_cr_arp_d_viewpnt?ie=UTF8&ASIN=B01H1I9X60#R2N2OQNG9UNUAG
>    So, it looks like adapter and/or its firmware is not fully compatible with OS’s other then Windows
> 5. If your main goal is to perform drive diagnostics on Mac - simply use another model of SATA-USB adapter/enclosure/dock.
> 
> P.S. Also, it is not fully clear why you have decided to manually "improve" the installation of the already installed driver. So, as a result, there is some possibility that you somehow incorrectly (re)installed driver and as a result, you now have outdated or corrupted kext/driver cache in macOS.
> When the macOS boots - it saves all currently used kexts (drivers) into a cache for faster booting next time - and if you move things around in “/Library/Extensions/” and “/System/Library/Extensions/” - you won't see those changes reflected until the caches are rebuilt.
> Type in the terminal following commands:
> 
> ```
> sudo kextcache -i /
> sudo kextcache -u /
> sudo kextcache -system-caches
> ```

Hello, I wrote you back in November but the issue I wrote you about had nothing to do with this 3rd party driver, but with crash which then never happened again so I didn't reply back (but I did receive that email).

However I am not sure if you received my emails I sent 2 weeks back (even now I replied to your email I received after contacting that payment wall) so I hope you received it now.

Did rebuild caches, but command "sudo kextcache -u /" is giving me: "Warning: /AppleInternal/Library/Extensions: No such file or directory" All other went just fine.

Please confirm that you received my email I sent you few hours back so we can continue communicating there, thank you (I am receiving all of your emails just fine in my inbox).