kasbert/OS-X-SAT-SMART-Driver #60: Installer 'Failed to Install'

ivanhass, 2020-04-03:
Hi,
I attempted to install SATSMARTdriver into my mac El Captain however i got an error message saying the installer failed. Screengrabs attached.

Any idea how to resolve this?

Thanks

alvarnell, 2020-04-03:
OS-X-SAT-SMART-Driver is no longer being supported for recent versions of macOS. You will need to obtain a signed copy from a 3rd Party Developer that has taken care of that, such as [DriveDx](https://binaryfruit.com/drivedx/usb-drive-support).

ivanhass, 2020-04-03:
Hi, 
I got the app from DriveDX who pointed me here for help on the Satsmart driver when it failed to install. My os isn't the most recent as its 10.11.6 (limited to due to other apps)

The satsmart driver installer came from DriveDX website (https://binaryfruit.com/drivedx/usb-drive-support). See at the bottom for support. I followed it and am now here...

alvarnell, 2020-04-03:
I see that Sierra didn't require signed kernel extensions, so you might want to try the ones from here, but that's all I can suggest.

I don't know why they would have sent you here. The SAT-SMART Developer hasn't been responding here for a very long time and about the only other persons, other than me, to post here has been BinaryFruit, the DriveDx developer.

ivanhass, 2020-04-03:
Thanks for the advice. I also tried the pre-compiled version 0.8 from GitHub and it had the same error. It offered to show me the error but then didn't show anything...

I will carry on hunting or give up and get SoftRaid Lite

ivanhass, 2020-04-03:
FYI - even though it said it failed to install, it actually did and I now have SMART on USB drives!

apple4ever, 2020-04-03:
You can also download [SMART Utility](https://www.volitans-software.com/apps/smart-utility/) which is compatibility back to 10.9 and includes a supported driver

alvarnell, 2020-04-03:
Glad it actually worked. I know I have it working in Sierra through Catalina and don’t recall having seen the error you saw, but back when I installed mine in El Capitan, I’m sure it was a different installer at the time.

If you are able to close the issue, I think that would help.