kasbert/OS-X-SAT-SMART-Driver #62: How could OS-X-SAT-SMART-Driver in future works under macOS 11 Big Sur

macuserguru, 2020-07-11:
I read that macOS 11 Big Sur don't like kext extensions.
How could OS-X-SAT-SMART-Driver works in future?

alvarnell, 2020-07-11:
First, it's not clear exactly when kernel drivers will no longer work. Apple has only said that they will need to be converted to system drivers in a future version of macOS. Latest rumors are that it won't be with the Fall introduction of Big Sur.

More importantly, development stopped a very long time ago, so any future changes will have to come from one of the third party utility developers that currently include working versions (e.g. SMART Utility, DriveDx, TechTool Pro, etc.).

BinaryFruit, 2020-07-13:
> I read that macOS 11 Big Sur don't like kext extensions.
> How could OS-X-SAT-SMART-Driver works in future?

Your information is not fully correct.
Not all kernel extensions will not work in macOS 11, but only a subset of them.
Apple clearly states that: "Kernel programming interfaces (KPIs) will be deprecated as alternatives become available".

At the moment, not all kernel extensions have alternatives, so they will work in macOS 11+ until Apples will provide an alternative KPIs for them. macOS 11 will continue to link and load kexts that have no System Extension solution available.

The types of kernel extensions that will not be supported are listed in Apple's [documentation](https://developer.apple.com/support/kernel-extensions/).

These are various network filters (for firewalls a-la Little Snitch) and drivers of custom USB
devices. But I want to note that this does not apply to generic external USB drives.
The SAT-SMART-driver used to enable diagnostics of external USB drives should work in macOS 11+

--
DriveDx Team / BinaryFruit

darikharian, 2020-07-14:
Hello master =)
I have recently installed your SAT SMART Driver in Mac OS Big Sur Beta 1, when connecting an external disk drive the system throws a kernel panic, I installed it to be able to use the DriveDx software which runs without problems. In order to be able to continue connecting external disk drives to my Macbook Pro without rebooting, sadly I have had to uninstall SAT SMART Driver.
I hope you are well, greetings!

alvarnell, 2020-07-15:
I suspect you will get a faster response by contacting the DriveDx Team directly as this driver is no longer being maintained on this site.

apple4ever, 2020-07-15:
I can confirm that the driver does not work in Big Sur with SMART Utility either. 

I'll be taking a look to see if I can fix it.

Smexhy, 2020-09-25:
I am using this driver with DriveDx diagnostic software for disks on macOS. On the latest beta 8, after opening DriveDx while my external HDD is connected and this plugin is installed, my mac almost immediately crashes and restarts. Hopefully this can be updated someday to work with big sur as I know they changed something about kernel extensions in this macOS version. It's really useful to have.

macuserguru, 2020-09-26:
Please - DriveDX should fork this driver and further develop them.
Why don't Apple include this feature in his OS?

macuserguru, 2020-09-26:
I would even pay some €/$ for it, as an additional DriveDX license.

alvarnell, 2020-09-26:
`DriveDX should fork this driver and further develop them.`
To some extent, they already have, in the they have signed it now. I would not be surprised if they were able to use the [System Extension and DriverKit]([url]https://developer.apple.com/system-extensions/(url)) to evolve compatibility should it continue to cause issues when Big Sur is released in a few weeks.

macuserguru, 2020-09-26:
I think we could wait for it. :-)

yamazaki-kiyoshi, 2020-10-05:
When I install the SAT SMART Driver on macOS Big Sur beta 9 and access an external USB HDD, the OS crashes. Is there any way to fix this problem?
I contacted DriveDx support last month and haven't received an answer yet.

alvarnell, 2020-10-05:
As I said above, SAT SMART is no longer in active development here, so you will have to wait and see if the folks at DriveDx can find a way to make it Big Sur compatible.

Gondomir, 2020-10-15:
> I contacted DriveDx support last month and haven't received an answer yet.

Same here. I think, there are no developers working at DriveDx anymore.

apple4ever, 2020-11-17:
The driver does work on Big Sur release, but you have to unplug all drives to get it to load.

xz4gb8, 2020-11-17:
Which version?

James R. Cutler
James.cutler@consultant.com
GPG keys: hkps://hkps.pool.sks-keyservers.net

alvarnell, 2020-11-18:
It would not be any version from this site. Would need to be one of the signed versions from DriveDx, SMART Utility or a few other 3rd party utilities that use it.

macuserguru, 2020-11-18:
> The driver does work on Big Sur release, but you have to unplug all drives to get it to load.

Which version from DriveDX?

apple4ever, 2020-11-18:
I use the driver in SMART Utility (since I develop that).

BinaryFruit, 2020-11-19:
> > The driver does work on Big Sur release, but you have to unplug all drives to get it to load.
> 
> Which version from DriveDX?

0.10.2 or 0.8.2 

Both versions are included in DriveDx 1.10.x and are also available for download on our [website](https://binaryfruit.com/drivedx/usb-drive-support).

macuserguru, 2020-11-19:
0.10.2 or 0.8.2 - which versions running under Big Sur and ARM CPUs?
Running kext still under Big Sur?

alvarnell, 2020-11-19:
Apparently either. Whichever one works best with your USB drives. Recommendation is to try 0.10.2 first and if it doesn't work with your setup try 0.8.2.

And yes, still has a .kext, but has been loading in /Library/Extensions rather than /System for awhile now.

-Al-

On Nov 18, 2020, at 23:49, macuserguru <notifications@github.com> wrote:
> 0.10.2 or 0.8.2 - which versions running under Big Sur and ARM CPUs?
> Running kext still under Big Sur?