kasbert/OS-X-SAT-SMART-Driver #63: Cannot install on Catalina

llamafilm, 2020-09-20:
I am trying to install on macOS 10.15.6 and I get the error, "This package is not compatible with this version of macOS. The package is trying to install content to the system volume."

[image]

llamafilm, 2020-09-20:
I was able to get this working by disabling SIP and manually installing it to my home folder, then copying to the Data volume.

- Reboot to Recovery mode with Cmd-R
- From Recovery mode, open a Terminal and type `csrutil disable`, then reboot
- `installer -verbose -pkg SATSMARTDriver-0.8-Release.pkg -target CurrentUserHomeDirectory`
- `sudo cp -r ~/System/Library/Extensions/SATSMART* /System/Volumes/Data/Library/Extensions/`

I would love to be able to run this with SIP enabled.

jdm0, 2020-09-20:
If memory serves, the install script is written to target `/System/Library` — which it shouldn't, and never should have, as everything under `/System` is reserved for official macOS stuff. The correct install target is `/Library` — that's where third-party stuff belongs. The install script needs to be corrected.

alvarnell, 2020-09-20:
The driver, including the installer is no longer being supported here by the original developer, so you need to use the one that has been so modified by the folks at Binary Fruit to install in the correct location:

<https://binaryfruit.com/drivedx/usb-drive-support>

-Al-

On Sep 20, 2020, at 11:56, jdm0 <notifications@github.com> wrote:
> If memory serves, the install script is written to target /System/Library — which it shouldn't, and never should have, as everything under /System is reserved for official macOS stuff. The correct [...]

porg, 2021-01-19:
A workaround for installation:
1. Open `SATSMARTDriver-0.10.dmg` with [SuspiciousPackage](https://www.mothersruin.com/software/SuspiciousPackage/)
2. Extract the folder "Extensions" with the two .kext files to a temporary location (i.e. `~/Desktop` or `~/Downloads`)
3. Manually move the two .kext files to `/Library/Extensions/`

Whether that creates a working kernel extension I cannot tell now as I currently do not have a USB enclosure around me to test. But at least you can get past the installation restriction.

porg, 2021-01-19:
As of today the newest version at BinaryFruit is version 0.10.2 with the KEXT files built on 2019-09-11 and the PKG file wrapping them being built on 2020-10-27.