kasbert/OS-X-SAT-SMART-Driver #64: Incompatible with macOS 11.0.1 and Apple Silicon Mac

dukai1011, 2020-11-24:
[image]

Please see the attach screenshot, both the 0.8, 0.10 and 0.11 version doesn't work on Apple silicon M1 processor Mac.

alvarnell, 2020-11-24:
Where did you obtain those copies? None of the ones on this site will ever work because they are unsigned. You must obtain them from DriveDx or SMART Utility or some other third party utility that signs those drivers.

dukai1011, 2020-11-24:
You can bypass the the identified developer by hold the ctrl key to run the application.

alvarnell, 2020-11-24:
Sure, but that won't allow the drivers to be used. There is no way to bypass the loading of an unsigned driver. Go to one of the utility sites I mentioned to download their signed versions for free.

macuserguru, 2020-12-31:
For future development.
Gives a chance for USB S.M.A.R.T. driver without using kext? And could work it with the new mechanism "system extensions" too?

alvarnell, 2020-12-31:
There has been no development effort here for years, so unless one of the drive utility developers or someone else wants to assume responsibility, there won't be any such future development.

macuserguru, 2020-12-31:
I know that's not a part of this project! 
My question was a general one.
Is it at all possible to get the S.M.A.R.T. values of a USB hard disks with the new feature "system extensions"?

closed 2020-12-31 by dukai1011