kasbert/OS-X-SAT-SMART-Driver #65: Disk Utility SMART Status "Not Supported", but DriveDx SMART Support "Enabled"

travelswagger, 2020-12-01:
Hello. Could you kindly assist? I'm running macOS High Sierra 10.13.6 with DriveDx 1.10.1. I have 2 Western Digital EasyStore WD080EZAZ-11TDBA0, both with multiple volumes. 

Is SAT Smart driver 0.10.2 installed and running correctly? 
- macOS Disk Utility shows each drive SMART Status to be "Not Supported" 
- however, DriveDx SMART Support indicates "Enabled"  
- furthermore, DriveDx Advanced SMART Status indicates "OK" 

Thank you for clarifying as I am not technical but wish to diagnose and monitor my external drives.

alvarnell, 2020-12-01:
Sounds like your driver is installed and running correctly.

SAT-SMART has not been supported here for several years now, so there's a good chance nobody will notice your request. Perhaps DriveDx or SMART Utilities can help. I know DriveDX posts about "driver v0.8.2 (try if v0.10.2 is not compatible with external drive)"

BinaryFruit, 2020-12-09:
Thanks for your interest in our software.
Please check our online FAQ - [Disk Utility reports SMART status as “Not Supported”, but DriveDx reports SMART support as “Enabled”](https://binaryfruit.com/drivedx/kb#apple-disk-utility-external-drive-smart).

alvarnell, 2020-12-09:
Note that the OP is running macOS High Sierra and your FAQ reverse to macOS Catalina and above.

BinaryFruit, 2020-12-09:
> Note that the OP is running macOS High Sierra and your FAQ applies to macOS Catalina and above.

Ok, thanks. It was relatively long ago, so we don't remember exactly the moment when these changes were included in macOS. Yes, most probably this happened in macOS 10.13.6 as [macOS 10.13.6 is the minimum OS required for Notarization](https://developer.apple.com/documentation/xcode/notarizing_macos_software_before_distribution).

BinaryFruit, 2020-12-09:
> Note that the OP is running macOS High Sierra and your FAQ applies to macOS Catalina and above.

Ok, thanks. It was relatively long ago, so we don't remember exactly the moment when these changes were included in macOS. Yes, most probably this happened in macOS 10.13.6 - as [macOS 10.13.6 is the minimum OS required for Notarization](https://developer.apple.com/documentation/xcode/notarizing_macos_software_before_distribution?language=objc#3087727).

BinaryFruit, 2020-12-09:
All in all, this issue related to the so-called "Hardened Runtime" feature of recent macOS versions. Since 10.13.6 macOS components do not load any non-Apple dylibs (like for example SATSMARTLib.plugin) and since Big Sur - macOS kernel does not load non-Apple kexts during system startup process (like for example SATSMARTDriver.kext).