kasbert/OS-X-SAT-SMART-Driver #69: Unable to remove SATSMARTDriver.kext

redvwyahoocom, 2021-04-09:
I have used the Terminal Command "sudo rm -r /Library/Extensions/SATSMARTDriver.kext" to remove SATSMARTDriver.kext, but receive the following error:   No such file or directory

Julien@juliens-imac ~ % sudo rm -r /Library/Extensions/SATSMARTDriver.kext
rm: /Library/Extensions/SATSMARTDriver.kext: No such file or directory
Julien@juliens-imac ~ % 

Screen shot below:

[image]
[image]

I have also tried to drag this item to the trash, but since the file is grayed out that does not work.

How do I remove the file SATSMARTDriver?

alvarnell, 2021-04-10:
I can't tell from your Find screenshot exactly where your SATSMART extensions are located or what version of macOS you are running, but until recently they were installed to /System/Library/Extensions/ which are SIP protected by the latest versions of macOS.

But if they are actually in /System/Library/Extensions/ in a SIP enabled macOS, then you must either disable SIP temporarily or remove them from Recovery.

redvwyahoocom, 2021-04-10:
Could you provide how to disable SIP?

It is located in the path you listed.

Julien Antkies
***@***.***

alvarnell, 2021-04-10:
First, let me just say that both methods will require you to boot into Recovery, so deleting them from there is faster but somewhat more dangerous if you happen to mistype a Terminal command.

Since Apple no longer publishes how to disable SIP, I'm reluctant to do so here, but a browser search will reveal several sources of printed and video instructions.

redvwyahoocom, 2021-04-10:
I found the command, thank you.

Would it leaving the file in place cause any issues?

Julien Antkies
***@***.***

alvarnell, 2021-04-10:
I'm not aware of any issues that it might cause. I suspect it's like most chuff that gets left behind and simply takes up some small amount of disk space.

redvwyahoocom, 2021-04-10:
Much appreciated.

closed 2021-04-10 by redvwyahoocom