kasbert/OS-X-SAT-SMART-Driver #70: Prompt to approve driver not showing up in OSX 10.14.6

soumdtt, 2021-06-18:
I have installed the SATSMART 0.10 but after installing it the prompt to approve the driver is not showing up in Security & Privacy pane in the System Preference. Is there an alternate way to approve the kext?

alvarnell, 2021-06-18:
If you used the installer here, then it will never work. You have to use an installer available from Smart Utility, DriveDX or one of the other utilities that include the revised versions that will work with recent macOS.

-Al-

soumdtt, 2021-06-18:
@alvarnell 
I am using the SATSMART downloaded from DriveDX only, but they suggested to report any problems with the driver in this forum.

alvarnell, 2021-06-19:
Well that's a disappointing cop out. Considering that SAT-SMART Driver is no longer supported here and when it was there was no need to approve it in Security & Privacy, this is close to the last place I would send anybody for such an issue. 

I take it that SAT-SMART isn't working for you with regard to a USB external drive. It's been at least a couple of years since I installed the DriveDX version in Mojave (or perhaps an earlier macOS) so I don't even recall if such approval was required at the time. In checking my Mojave setup today I see that I have a signed 0.10 driver installed in /System/Library/Extensions, /Library/Extensions and /Library/StagedExtensions/Library/Extensions. I'm certain that the first is left over from an earlier installation when installing in /System by an admin was still allowed. Just the other day I cleaned out my extensions to get rid of any that were no longer needed using `sudo kextcache --clear-staging` and rebooting to rebuild the cache, so the /Library portion should be up-to-date.

Recommend you check the Console for any error messages referencing a kext... process in conjunction with SAT-SMART.  Assuming that Drivers are treated the same as Kernel Extensions and Mojave behaves the way High Sierra did, this Tech Note might be of use: https://developer.apple.com/library/archive/technotes/tn2459/_index.html.