kasbert/OS-X-SAT-SMART-Driver #76: Booting from external?

offworldOwl, 2022-04-08:
Could SAT-SMART be enhanced to work when booting from external drive in macOS 12?

This used to work in Mojave, but...

alvarnell, 2022-04-09:
SAT-SMART has been unsupported here for a very long time now. You can possibly get answers from the developers of DriveDX or SMART Utility which have been finding ways to make it useful in recent macOS versions.

macuserguru, 2022-04-09:
I use an M1 MacMini with Monterey 12.3.1 - if I let the external TM 2.5" harddisk connected after shutdown, the S.M.A.R.T. driver works after new boot without reconnect my external harddisk.
I use the last SATSMARTDriver-0.10.3.macOS11_and_AppleSilicon.zip from Binaryfriut.
https://binaryfruit.com/drivedx/usb-drive-support

alvarnell, 2022-04-09:
I don't recall DriveDx ever being available in the Mac App Store, but I purchased my copy many years ago on their site and kept it automatically updated.
I would think it more appropriate to contact BinaryFruit Support directly at https://binaryfruit.com/drivedx/support. I haven't tried it in Monterey yet, but I also haven't heard from any users that have there being any problems with it.

macuserguru, 2022-04-09:
> I don't recall DriveDx ever being available in the Mac App Store, but I purchased my copy many years ago on their site and kept it automatically updated. I would think it more appropriate to contact [...]

Please look at your mailserver - a direct mail and a message on the support pages returned an error message:

> MailEnable: Message could not be delivered to some recipients.
The following recipient(s) could not be reached:

	Recipient: [SMTP:[support@binaryfruit.com](mailto:support@binaryfruit.com)]
	Reason: Mail server for [binaryfruit.com](http://binaryfruit.com/) is not contactable. Message has expired and has been returned to Sender.

or

> The mail system

<[binaryfruit.info@gmail.com](mailto:binaryfruit.info@gmail.com)>: host [gmail-smtp-in.l.google.com](http://gmail-smtp-in.l.google.com/)[142.250.114.27]
   said: 550-5.7.26 This message does not have authentication information or
   fails to 550-5.7.26 pass authentication checks. To best protect our users
   from spam, the 550-5.7.26 message has been blocked

offworldOwl, 2022-04-09:
I'm so glad this conversation caught attention.

I'm 51-year old college drop out with no experience on coding.
But I'm willing to learn, if that's what it takes to monitor your storage in mac ecosystem.

AFAIK, sat-smart is the ONLY way to monitor external drives' health in macOS.
If there's an alternative, I'd like to know.

My biggest grief is mac's non-removable storage these days.
I won't do what Apple tells me, that I should buy biggest storage with their prices, because I might need it sometime in 5 years in the future.
To my mind Apple behaves just like hp, with biggest revenues from ink. Or some company selling sugared water.

So, I bought a mini at the beginning of 2019 with small storage, with the idea of running OS from internal and the rest from external. Back in the days, when mac desktops had multiple INTERNAL storage's this was pretty normal. Then I learned that macOS isn't so much "regular unix" anymore with Apple's multiple undocumented security layers sliced and diced within it.

Because of that, I use solely ONLY external drives to boot from with my mini. And there are so many problems with it, that I won't tell if not asked.
With the new restrictions that kext can be loaded only at the time of boot (is it really? Even when SIP is disabled?), I have not been able to boot from external with sat-smart loaded at the boot. Trying that leads to 3-4 hour bootloop and after that the OS disables the ket automatically. Since the bootloop takes so much time, I haven't been able to test it a lot.

I did send a mail to Binaryfruit in February.
Didn't get any error message, but got no answer either.

br,
toke

ps. Kasbert! Jos luet näitä, niin huikkaatko?

macuserguru, 2022-04-09:
DriveDX and other similar have no problems to catch the S.M.A.R.T. values of Apples new build in SSD's of the M1 Macs.
A benefit of this SSD's - they are very fast (high data rate) - but expensive.
I spend my M1 MacMini the max available configuration - 16 GB RAM, 2 TB SSD

offworldOwl, 2022-04-09:
This is widely known.
Does not help with my mini with 256GB internal ssd.
Apple should use more often that “option” key on their keyboards.

I would accept hand down less security and slower speed, if there were DESKTOP macs with upgradeable RAM and storage with REASONABLE price.

I’m not sure if I should repeat: I boot ONLY from external. After Monterey, I can’t check my bootdrive’s health anymore.
With Mojave I could.
Haven’t tested the OSses between. I’m out of this yearly upgrade madness.

macuserguru, 2022-04-09:
On which technology runs the external drive? USB3 / USB-C or Thunderbolt?
AFAIK Thunderbolt drives don't need S.M.A.R.T. drivers, but I can't test this self.
Which manufacturer and model of external drive use you.?

macuserguru, 2022-04-09:
AFAIK - Loading S.M.A.R.T. .kext drivers in boot cycle you must decrease the macOS security.
Did you want this?

offworldOwl, 2022-04-10:
Yes I do.
I don’t want any additional security that Apple forces me to have, that simultaneously adds Apple Tax to the whole new level.

I’ve used 10G-usb enclosures and dongles. The barnds of these 3 are irrelevant, IMHO.

I haven’t bought any tb-enclosures, since they are way overpriced and does not offer pretty much no advantage over cheaper usb enclosures, since I’m using “normal” sata drives. They are fast enough for me.

I can’t think any logical reason why tb-attached drive would be considered as internal drive and therefore smartmontools could read it’s storage’s smart data.

What I am thinking about, is that if with OpenCore I could fake my boot drive to be “internal”...

Bottacco, 2023-03-08:
> … I’ve used 10G-usb enclosures and dongles. The brands of these 3 are irrelevant, IMHO.

Well, not specifically the brands, but the SATA controller that they use is very relevant. Many external enclosures use a SATA controller that doesn't support SMART passthrough (or SAT for that matter), so installing this driver won't help you. That is why the "brand" sometimes gives you clues if the enclosure is supported or not.

>I haven’t bought any tb-enclosures, since they are way overpriced and does not offer pretty much any advantage over cheaper usb enclosures, since I’m using “normal” sata drives. They are fast enough for me. I can’t think any logical reason why tb-attached drive would be considered as internal drive and therefore smartmontools could read it’s storage’s smart data.

Thunderbolt enclosures are "presented" to the operating system as if they were connected directly to the PCIe bus, so they give the full control of the storage device like if it were connected internally. That is the real advantage of Thunderbolt. I have the original OWC Drive Dock (Thunderbolt 2 version) so I can pop in any SATA drive and have control of it with smartmontools or any other utility without having to use the SATSMART driver. That is the logical reason why Thunderbolt is better that USB for storage devices.

Lately there are some new USB-C 3.2 enclosures that do a passthrough of the internal drive to the operating system, but you need a modern Mac with native USB-C ports (if you use a converter from USB-C to USB-A to connect it to an older Mac it will mount the disk and it will work as a storage device, but you won't be able to use smartmontools or any other utility that needs SAT).

Now that USB4 includes the Thunderbolt protocol and unifies everything we will have to wait and see what Apple and storage manufacturers do. But I hope that SAT is standard in any new enclosure with the USB4 protocol stack, although the future is uncertain because there are subsets of the protocol. Time will tell.

>What I am thinking about, is that if with OpenCore I could fake my boot drive to be “internal”...

Using OpenCore won't help you. The disk will appear as an internal drive to the operating system, but the drive won't change its current behavior as the loading process on startup is at the physical layer. The OpenCore trick to show a disk as internal is just to not eject the drive while going to sleep and to show the icon of the drive as an internal drive.

I hope this clarifies a bit the use of external drives. But in any case, if you want to boot always from an external drive you will have to lower the default/standard security Apple ships.