kasbert/OS-X-SAT-SMART-Driver #77: Need arm64e on M1 device

trK54Ylmz, 2022-07-14:
I have installed BinaryFruit's SAT SMART Driver, when I run `diskutil info`, I got the error below,

```
diskutil[25583:146119] Error loading /Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib:  dlopen(/Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib, 0x0106): tried: '/Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib' (fat file, but missing compatible architecture (have 'x86_64,arm64', need 'arm64e'))
```

Shorter version,

```
(fat file, but missing compatible architecture (have 'x86_64,arm64', need 'arm64e'))
```

I think the kext does not support arm64e architecture. Is anyone came across with this error?

The device is Mac Mini 2020.

macuserguru, 2022-07-15:
I installed from https://binaryfruit.com/drivedx/usb-drive-support
this version 'for macOS 11 and Apple Silicon' 
and it works fine on M1 MacMini Big Sur an now Monterey 12.4

trK54Ylmz, 2022-07-15:
Hi @macuserguru, I have installed the same exact driver on Mac Mini, Monterey 12.4 but Disk Utility says `S.M.A.R.T. status: Not Supported.`.

On `diskutil`, I always get `fat file, but missing compatible architecture` error.

```bash
$ lipo -info /Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib
Architectures in the fat file: /Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib are: x86_64 arm64
```

macuserguru, 2022-07-15:
Delete all kext as you can read here, reboot and install then new
https://binaryfruit.com/drivedx/usb-drive-support

Run both versions of rm to remove older versions - and search for certainty for SATSMARTDriver.kext and SATSMARTLib.plugin in your system after you have delete. 

UNINSTALL INSTRUCTIONS
    On macOS 10.11+ – type following commands (in Terminal.app):
    sudo rm -r /Library/Extensions/SATSMARTDriver.kext
    sudo rm -r /Library/Extensions/SATSMARTLib.plugin
    Reboot computer

    On old versions of macOS – type following commands (in Terminal.app):
    sudo rm -r /System/Library/Extensions/SATSMARTDriver.kext
    sudo rm -r /System/Library/Extensions/SATSMARTLib.plugin
    Reboot computer

trK54Ylmz, 2022-07-15:
I have removed the files, and rebooted the computer.

After reboot,

```bash
$ kmutil showloaded|grep "binaryfruit"
No variant specified, falling back to release
```

When I run `diskutil info diskX`, I got no error (SMART Status: Not Supported)

Then I have installed new file which I have downloaded (from "for macOS 11 and Apple Silicon    ") and rebooted again.

```bash
$ kmutil showloaded|grep "binaryfruit"                     
No variant specified, falling back to release
  238    0 0xfffffe0006f187f0 0x4000     0x4000     com.binaryfruit.driver.SATSMARTDriver (0.10.3) ....
```

The error stays same when I run `diskutil info`,

```
diskutil[1766:12463] Error loading /Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib:  dlopen(/Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib, 0x0106): tried: '/Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib' (fat file, but missing compatible architecture (have 'x86_64,arm64', need 'arm64e'))
```

macuserguru, 2022-07-15:
What are your settings in
system settings / security / allow apps download from?

I use 'App Store and verified developers'

macuserguru, 2022-07-15:
All the same
https://macsupport.tuxera.com/hc/en-gb/articles/4409208805522-How-do-I-enable-third-party-kernel-extensions-on-Apple-Silicon-based-Macs-

macuserguru, 2022-07-15:
lipo -info /Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib

2022-07-15 19:07:09.145 xcodebuild[1924:19065] Requested but did not find extension point with identifier Xcode.IDEKit.ExtensionSentinelHostApplications for extension Xcode.DebuggerFoundation.AppExtensionHosts.watchOS of plug-in com.apple.dt.IDEWatchSupportCore
2022-07-15 19:07:09.145 xcodebuild[1924:19065] Requested but did not find extension point with identifier Xcode.IDEKit.ExtensionPointIdentifierToBundleIdentifier for extension Xcode.DebuggerFoundation.AppExtensionToBundleIdentifierMap.watchOS of plug-in com.apple.dt.IDEWatchSupportCore
Architectures in the fat file: /Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib are: x86_64 arm64

macuserguru, 2022-07-15:
Did you see USB HD Drives in DriveDX?

trK54Ylmz, 2022-07-15:
> lipo -info /Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib
>
> 2022-07-15 19:07:09.145 xcodebuild[1924:19065] Requested but did not find extension point with identifier [...]

I got the same thing but I don't understand why you don't get the error because the plugin did not compile with `arm64e` architecture,

```
$ lipo -info /Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib

2022-07-15 20:10:03.932 xcodebuild[3260:34940] Requested but did not find extension point with identifier Xcode.IDEKit.ExtensionSentinelHostApplications for extension Xcode.DebuggerFoundation.AppExtensionHosts.watchOS of plug-in com.apple.dt.IDEWatchSupportCore
2022-07-15 20:10:03.932 xcodebuild[3260:34940] Requested but did not find extension point with identifier Xcode.IDEKit.ExtensionPointIdentifierToBundleIdentifier for extension Xcode.DebuggerFoundation.AppExtensionToBundleIdentifierMap.watchOS of plug-in com.apple.dt.IDEWatchSupportCore
Architectures in the fat file: /Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib are: x86_64 arm64
```

For example,

```bash
$ lipo -info /System/Library/Extensions/IOAccelerator2D.plugin/Contents/MacOS/IOAccelerator2D
Architectures in the fat file: /System/Library/Extensions/IOAccelerator2D.plugin/Contents/MacOS/IOAccelerator2D are: x86_64 arm64e
```

The `IOAccelerator2D.plugin` compiled as `arm64e` as well.

From Apple [documentation](https://developer.apple.com/documentation/kernel/implementing_drivers_system_extensions_and_kexts),

```
Kexts on Apple silicon must support the arm64e architecture
```

trK54Ylmz, 2022-07-15:
> Did you see USB HD Drives in DriveDX?

Yes, I can see on DriveDX, but on Disk Utility it says does not support and `diskutil` tool has problems also,

[image]

macuserguru, 2022-07-15:
Download the new DriveDX.app from https://binaryfruit.com/drivedx - this include a SATSMART too.

macuserguru, 2022-07-16:
I use DriveDX 1.11.0 direct downloaded from https://binaryfruit.com/drivedx
The other version from Appstore are ver. 1.7.0

In the DriveDX 1.11.0 app-package are included two SMART driver 0.10 versions

[image]

macuserguru, 2022-07-16:
> Actually, I already installed the application from that link and the driver was in it. It didn't work. In my humble opinion, the `SATSMARTLib.plugin` should compile from source code with `arm64e` as target architecture.

Delete all this drivers, reboot, download this version '**for macOS 11 and Apple Silicon**' and install this
https://binaryfruit.com/download/mac/satsmartdriver/SATSMARTDriver-0.10.3.macOS11_and_AppleSilicon.zip
and it works fine on M1 MacMini Big Sur an now Monterey 12.4

macuserguru, 2022-07-16:
Look for registrated .kext .dext
Thanks to 'Apfeltalk'
https://www.apfeltalk.de/community/threads/sicherheit-ausnahme-kext-dext-erteilen-auch-in-terminal-moeglich.568728/#post-5709790
....

`sudo /usr/sbin/klist_cdhashes /var/db/SystemPolicyConfiguration/KextPolicy /tmp/KextPolicy.plist`

trK54Ylmz, 2022-07-16:
I believe `SATSMARTDriver` has no problem but `SATSMARTLib` has. I have tried the command as well. In my humble opinion, the problem is related to the arm64 build. On my Intel Mac, S.M.A.R.T. is verified, Disk Utility CLI works also.

I think you should see error message when you run `diskutil info disk[disk ID]` on your M1 Mac.

macuserguru, 2022-07-17:
`Protocol:                  USB
2022-07-17 16:08:04.456 diskutil[6038:244182] Error loading /Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib:  dlopen(/Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib, 0x0106): tried: '/Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib' (fat file, but missing compatible architecture (have 'x86_64,arm64', need 'arm64e'))
`

But this is irrelevant for the display of the USB hard disk in DriveDX!

Or with SmartReporter:
SmartReporter: S.M.A.R.T.-checked hard disks: Hardware, iSCPreboot, xART, Update, Preboot, VM, Macintosh HD (........) WD 5TB 3rd (WD-.......)

With both DriveDX and SmartReporter I could see S.M.A.R.T. values from external hard-disks.