kasbert/OS-X-SAT-SMART-Driver #81: Popular Enclosure

laevda, 2023-02-24:
Dear developer, please add support for the following boxes in combination with SSD drives!
 
Enclosure model: JEYI i9 | based on JMS 583. SSD – NVME M.2 Samsung PM9A1 (OEM-version 980 PRO)
  |   "PassThroughMode" = "none"
  |   "Enclosure Name" = "Unknown 152d:0583"
  |   "SATSMARTCapable" = No
  |   "KnownEnclosure" = No

 Enclosure model: ORICO PWM2-G2 | Based on RTL 9210B. SSD – NVME M.2 WD Black SN770
  |   "PassThroughMode" = "sat12"
  |   "SATSMARTCapable" = Yes
  |   "Enclosure Name" = "Unknown 0bda:9210"
  |   "KnownEnclosure" = No

alvarnell, 2023-02-24:
SAT-SMART has not been under active development for over seven years now. Afraid you are on your own here.

macuserguru, 2023-02-24:
It's a great pity that this is not being developed further.
This would require a team that is also allowed from Apple to sign kext!
Or Apple itself would take over - would be even better!
The DriveDX developer (with whom I am in contact from time to time) offers signed versions, but only takes over the code 1:1. He has to make DriveDX Apple ARM ready himself, it still runs in Intel --> ARM emulation.

taprobane99, 2023-05-04:
Has anyone had any success with Jmicron or Realtek enclosures using a Windows VM (i.e. using Parallels, VMFusion, UTM)?

On a real Windows computer both work fine with CrystalDiskMark or smartmontools.

taprobane99, 2023-05-30:
how does CrystalDiskMark manage to read almost all usb enclosures?