kasbert/OS-X-SAT-SMART-Driver #82: No attached devices shown (hdiutil info)

wq9578, 2023-03-19:
The driver works well, but now (maybe after reboot) `hdiutil info` returns an empty list, not showing the attached devices:
```
framework       : 623.100.1
driver          : 623.100.1
```

Also, entering `diskutil info -all` produces the following error messages in between the regular output before each line starting with `SMART Status:              Not Supported` (similar to https://github.com/kasbert/OS-X-SAT-SMART-Driver/issues/71 and https://github.com/kasbert/OS-X-SAT-SMART-Driver/issues/77):
```
2023-03-19 18:46:01.400 diskutil[1277:18413] Error loading /Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib:  dlopen(/Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib, 0x0106): tried: '/Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib' (code signature in <B0BE47D9-02D0-35FF-950F-E3A83F32E147> '/Library/Extensions/SATSMARTLib.plugin/Contents/MacOS/SATSMARTLib' not valid for use in process: mapping process is a platform binary, but mapped file is not)
```

Nevertheless the devices supports SMART (e.g., `smartctl -H /dev/disk2`):
```
smartctl 7.3 2022-02-28 r5338 [Darwin 21.6.0 x86_64] (local build)
Copyright (C) 2002-22, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED
```

Used the installer from https://binaryfruit.com/download/mac/satsmartdriver/SATSMARTDriver-0.10.3.macOS11_and_AppleSilicon.zip, linked at https://binaryfruit.com/drivedx/usb-drive-support.

Apple M1 chip, recent macOS.

alvarnell, 2023-03-20:
Note that OS X SAT SMART Driver has appeared to be unsupported here for many years now. Binary Fruit provides signature support, but as far as I can determine, that is the extent of their effort.

killerovsky, 2023-06-19:
Error message "'SATSMARTLib' not valid for use in process: mapping process is a platform binary, but mapped file is not" is pretty self-explanatory. DiskUtility is an integral part of macOS ("platform binary") but any third-party driver (kext) is not. This is part of Apple System Integrity Protection since High Sierra 10.13.6. 
Why does nobody bother to check FAQs before posting something? Please check the online FAQ - https://binaryfruit.com/drivedx/kb#apple-disk-utility-does-not-use-sat-smart-driver