kasbert/OS-X-SAT-SMART-Driver #86: Samsung T7 SSD Not Recognized

verax-satya, 2023-07-07:
Hi! I am using this software with DiskDX and it looks like the Samsung T7 SSD is not compatible with this SAT SMART driver. Any solutions?
[image]

alvarnell, 2023-07-08:
I think you stand a better chance of getting an answer from DriveDx since the original developer appears to have abandoned supporting it here for several years now.

macuserguru, 2023-07-08:
All supported vendors are seen in 
/Library/Extensions/SATSMARTDriver.kext/Contents/Info.plist (show content)
I think, new vendors, new hard disk types must be append here, too.
SATSMARTDriver is signed, but I don't know if changing the info.plist requires re-signing.

The next adjustment is required in SMART tool itself.

macuserguru, 2023-07-08:
More about see smartmontools
https://github.com/smartmontools/smartmontools/releases 
You could install smartmontools (gives an dmg) and update their database too.
smartmontools works with SATSMARTDriver from binaryfruit (DriveDX)

destroy314, 2025-02-22:
Looks like this driver don't support that nvme ssd: https://binaryfruit.com/drivedx/kb#NVMe-SMART-over-USB
The only way might be use samsung magician app

macuserguru, 2025-02-22:
Installs samsung magician app a kext too?

destroy314, 2025-02-22:
I guess the app installer will handle it automatically, because I was asked to reboot during installation. Also, [here](https://forums.macrumors.com/threads/still-no-way-to-access-smart-information-of-external-nvme-drive-in-an-usb-enclosure-update-there-is-a-way.2383604/) says that SMART data can be accessed through a virtual machine. I tried using Wine 10.0 with CrystalDiskInfo, but it couldn’t find the hard drive.

macuserguru, 2025-02-22:
Please take a look witch kext are installed?
`kextstat | grep -v com.apple`
.
`% kextstat | grep -v com.apple
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
  245    0 0                  0x12f0     0x12f0     com.binaryfruit.driver.SATSMARTDriver (0.10.3) 241BAD32-9242-3B68-A291-314894D66D2D <54 53 20 4 3>`

destroy314, 2025-02-22:
Unfortunately I don't know

macuserguru, 2025-02-22:
> Unfortunately I don't know

Please run this command on your Mac if you installed any kext (or not).

`kextstat | grep -v com.apple`

kiler129, 2025-04-25:
From what I was able to put together, the only method of obtaining SMART for T7 on macOS is actually installing [Samsung Magician](https://www.samsung.com/us/memory-storage/magician-software/) **and** activating its kernel extension. This, by default, requires lowering security settings to allow kexts at all as Apple is phasing them out (see https://support.apple.com/guide/mac-help/change-security-settings-startup-disk-a-mac-mchl768f7291/mac). Few restarts later:

```
% kextstat | grep -v com.apple
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
  258    0 0                  0x6e9      0x6e9      com.samsung.magicianpssd.driver (8.3.0) 1F3D122F-04D6-3AE0-A3A0-9EF7267425DB <55 54 5 4 3>
```

After this Samsung Magician is able to manage the drive, pull SMART, update firmware etc. **This driver doesn't seem to support T7**. The drive [appears to be problematic even under Linux](https://www.claudiokuenzler.com/blog/1368/how-to-read-smart-drive-health-status-samsung-portable-ssd-external-drive) - despite using standard UASP for data, it doesn't implement SCSI over USB fully. On the newest firmware my Linux box is (again) not able to pull all information, even using different `-d` types.