kasbert/OS-X-SAT-SMART-Driver #87: SAT SMART Driver .10 in macOS Ventura 10.4.1(c) Prevents Boot

smith4562k3, 2023-07-23:
Installed the .10 version of this driver for external USB device support for use with DriveDx v1.12.0 (Build 750). As expected, System Settings/Security & Privacy prompted to 'Allow' the kernel extension to load, then rebooted my 2018 Intel Mac Mini...no love.

Within the first third of the boot process the Mini spontaneously rebooted and began to loop the same behavior. Safe boot was successful but simply deleting the two kernel extensions/driver files did not resolve the boot problem. Booting to Restore and running this command corrected the boot problem: kmutil trigger-panic-medic --volume-root /Volumes/Macintosh\ HD/

NOTE: The boot volume name in the command is assumed to be "Macintosh HD". Please use your actual volume name for the command to properly reset all 3rd party kernel extensions.

Hoping there's an updated driver coming to at least prevent the boot panic and loop behavior. Thanks.

kasbert, 2023-07-23:
Did you try rebooting with external drive disconnected?

alvarnell, 2023-07-24:
SAT-SMART Driver has not been supported on this site for several years now, so there won't be any new versions from this developer. Best to check with BinaryFruit on your issue. I've found their FAQs useful in dealing with most issues: https://binaryfruit.com/drivedx/kb.