kasbert/OS-X-SAT-SMART-Driver #89: LaCie 2Big Triple Firewire not mounting when SAT SMART driver installed

ivanhass, 2024-04-09:
Hi,
I have a LaCie 2big triple drive connected via firewire which will not mount when the SAT Smart driver is installed.
MacBook Pro 13.6.6, connected using usb-c to thunderbolt2 adaptor and then thunderbolt2 to firewire 800 adaptor.

I was directed here by DriveDX website

Greatful if you could advise?
Ivan

nb I also have a Lacie 12TB Thunderbolt2 connected via Thunderbolt2 which has a similar issue and crashes the computer if DriveDX is run. Other drives work fine.

kasbert, 2024-04-10:
I am afraid there is no fix for LaCie external drives. Uninstall the driver.

ivanhass, 2024-04-10:
Bummer !My observation is that it works fine for the LaCie 2BIG quad drive (the one with eSATA port) but not for the older LaCie 2BIG triple. The triple has FireWire 800 400 and USB. Oddly it did used to work with the older driver so I’m wondering if it was removed from the newer driver?On the plus side I run a mixed environment and love the support on WD Duo drives which now I’ve lost as I’ve had to uninstall.