kasbert/OS-X-SAT-SMART-Driver #90: Upgrade to Sequoia 15.0

macuserguru, 2024-09-22:
I have no problems after upgrade to 15.0
OS-X-SAT-SMART-Driver works well, too. Download from DriveDX homepage, signed & compiled.
https://binaryfruit.com/drivedx/usb-drive-support
Runs on an M1 Mac Mini.

voghuser, 2024-11-18:
I have a very peculiar and probably rare problem: SATSMART driver on a Mac mini M2 Pro with 15.1 (skipped 15.0) prevents Scubapro/UWATEC dive computers G2 and G2TEK to show their flash disk. No such problem on an Intel iMac. 
This flash disk is supposed to work like an USB stick, MS-DOS (FAT32), contains all data of the dive computer and is needed among other things for updates of the computer. This worked nicely for many years (7 years with the G2), various MacOS versions and SATSMART used for real USB hard disks. I. e. SATSMART ist by no means necessary for the dive computer. But now with MacOS 15 (15.1) the flash disk does not show up any more on the mini M2 Pro. However, it does show up on an iMac 2017 (Intel) with SATSMART and 15.1. When I delete SATSMART on the mini the flash disk shows up without any problems, apart from the usual problems with USB sticks and 15.1 (e.g. you have to wait for two minutes or so until size and free space become different from 0 in a finder window and you my write on the stick).
Yes, I confirm I have no problem with SATSMART and DriveDX.
So for my USB disks and DriveDX I need SATSMART, but it seems to keep the flash disk of the dive computer outside. Any ideas how to solve this problem? (I can use the old iMac for the time being, but ...)

ferdnyc, 2024-12-09:
@voghuser I don't have any help to offer with your issue, but you should _definitely_ be posting it as a new issue of your own, not as a comment on @macuserguru 's report about successfully running DriveDX on macOS 15. There's almost no chance your issue will get any attention, where it's currently posted.

(And you should probably be posting updates as new, separate comments, not by editing them into your one bigmclargehuge comment. Because it's turned into a wall of text that's growing increasingly harder to read.)

Blank lines between paragraphs would help a lot.

AlecKinnear, 2025-08-01:
For those with Apple Silicon, I've just successfully installed WD Drive Utilities which includes a basic SMART check via USB for WD drive enclosures (MyBook in this case), as well as both a quick test and an extended test. This is basically the DriveDX features without the granularity (one doesn't see On hours for instance or defective sectors).

I'm a big DriveDX fan but I've had too much trouble trying to get SatSmart Driver to work with DriveDX on my Apple Mx Macs.

macuserguru, 2025-08-02:
“WD Drive Utilities“, but I use Seagate/LaCie, Crucial Micron (USB SSD) drives in addition to WD drives.
Look at [https://github.com/smartmontools/smartmontools/issues/373](url)

For Crucial Micron SSD - in  DriveDX I could not see the Crucial X10 SSD drive, in 
SMARTReporter / Version:  3.2.0, Build: 2576, CoreCode
and
SMART Utility, Version 3.2.7 (3C142), ©2022 Volitans Software
yet.