kasbert/OS-X-SAT-SMART-Driver #91: "Diagnostics of external drives is unavailable" for DriveDx on MacBook Pro (Retina) w MacOS 12.7.6

cyberzen11, 2025-03-20:
I've tried multiple times to install the "SAT SMART Driver for DriveDx", removing prior SATSMART extensions from Library/Extensions (none get installed in System/Library/Extensions) before each new attempt as well as unplugging all external drives and rebooting the MacBook Pro.

Thus far I have not been able to get DriveDx to recognize the S.M.A.R.T. status of any external drive.

Is there a known issue with the driver and Monterey?  What else can I try?

Thanks!

crispybegs, 2026-02-26:
did you ever solve this? i can't install any available version on monterey

crispybegs, 2026-02-26:
thanks @arnstein99 . later on after posting that I actually found a working download link on the DriveDX site, here - https://binaryfruit.com/drivedx/usb-drive-support